EmbeddedRPC/erpc #5: C++ version of erpc_c files

ablaylock, 2017-10-04:
What version of c++ should the files in the erpc_c directory use? Some of the code requires c++11 for building erpcgen, is this also a reasonable requirement for the actual implementation?

Hadatko, 2017-10-05:
Hi,
currently we are not forcing to use c++11 version. Some embedded IDE are not supporting this standard, or they just a start support it. Currently we are aligned with C99.

ablaylock, 2017-10-05:
I am more specifically interested in the c++ portion of the infrastructure, not the c API which I can understand being C99. What version of c++ do the c++ files need to comply with? c++98, c++03, or c++11?

Hadatko, 2017-10-05:
Sorry, i was mixing two different things. Currently we are aligned with C++03.

closed 2017-10-09 by ablaylock