EmbeddedRPC/erpc #8: Inter-"version" compatability

ablaylock, 2017-10-09:
I am interested in having inter-"version" compatibility. 

It would be nice if mismatched clients and servers could talk. As it stands on the dev branch a version CRC is embedded prevents any interoperability between builds of IDL files that may have only added functionality but not removed it.

Hadatko, 2017-10-09:
Generating CRC should be optional in future. Although you don't need to use it. You can modify client/server setup functions in way that they will set some default CRC number.

ablaylock, 2017-10-09:
How should generating CRC's be optional? Should the erpc_client_init and erpc_server_init functions take a "erpc_crc_t object" so that the user can supply one? 

Another point would be having the erpcgen provide some additional guarantees on compatibility. What does a developer need to do to keep the serviceId, methodId, etc compatible? I wasn't able to find anything about that in the documentation? My best guess is the @id annotation but there are no examples and the discussion is limited.

Hadatko, 2017-10-09:
We are planning use annotation @crc when we will want use generated CRC. And then there will be probably added setup function to set crc value.

Yes with @id you can specify functions(method id) and interfaces(service id) ids so they will be always the same (for example for case when you change order of functions, or add some function declaration somewhere in middle of IDL file).