EmbeddedRPC/erpc #10: Build Error

CPPZombie, 2017-11-23:
Any suggestion if i have missed some steps before running 'make'?

System:
$ lscpu
Architecture:          x86_64
CPU op-mode(s):        32-bit, 64-bit
Byte Order:            Little Endian
CPU(s):                4
On-line CPU(s) list:   0-3
Thread(s) per core:    2
Core(s) per socket:    2
Socket(s):             1
NUMA node(s):          1
Vendor ID:             GenuineIntel
CPU family:            6
Model:                 69
Model name:            Intel(R) Core(TM) i7-4510U CPU @ 2.00GHz
Stepping:              1
CPU MHz:               944.734
CPU max MHz:           3100,0000
CPU min MHz:           800,0000
BogoMIPS:              5187.74
Virtualization:        VT-x
L1d cache:             32K
L1i cache:             32K
L2 cache:              256K
L3 cache:              4096K
NUMA node0 CPU(s):     0-3
Flags:                 fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm epb tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid xsaveopt dtherm ida arat pln pts

OS:
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.3 LTS
Release:	16.04
Codename:	xenial

Steps:
$git clone https://github.com/DusanCervenkaNXP/erpc.git
$cd erpc/
$make

result:
~/myrepo/erpc/erpc$ make
Generating erpcgen/src/erpcgen_parser.y
make[1]: /usr/bin/bison: Command not found
Makefile:138: recipe for target '/home/myhome/myrepo/erpc/erpc/Debug/Linux/erpcgen/obj/erpcgen_parser.tab.hpp' failed
make[1]: *** [/home/myhome/myrepo/erpc/erpc/Debug/Linux/erpcgen/obj/erpcgen_parser.tab.hpp] Error 127
/home/myhome/myrepo/erpc/erpc/mk/subdirs.mk:41: recipe for target 'erpcgen' failed
make: *** [erpcgen] Error 2

Hadatko, 2017-11-23:
Hi @CPPZombie , please read a README on home page: https://github.com/EmbeddedRPC/erpc
There are described dependencies for application. This error is saying that you are missing 'bison' app (sudo apt-get install bison). But there will be many more missings.

CPPZombie, 2017-11-23:
Thanks for the quick reply. I'll look into the pre compiled app.

Hadatko, 2017-11-23:
You are welcome.