EmbeddedRPC/erpc #11: Simple hello-world server/client

CPPZombie, 2017-11-28:
I have looked at the erpc_transport_setup.h and wonder if there are a call (maybe in other file) where you can set an IP-addr and Port for a server that a client from another machine can use to communicate to server, similar to gRPC?

Hadatko, 2017-11-28:
Hi, currently we don't have setup function for TCP transport if you mean that, but TCP transport is supported. You can find https://github.com/EmbeddedRPC/erpc/blob/master/erpc_c/transports/tcp_transport.h . If you want you can created setup function similar to other transport types.
This transport is used here:
https://github.com/EmbeddedRPC/erpc/blob/master/test/common/unit_test_tcp_client.cpp
https://github.com/EmbeddedRPC/erpc/blob/master/test/common/unit_test_tcp_server.cpp

CPPZombie, 2017-11-28:
Thank you, i'll check them out.

CPPZombie, 2017-12-06:
Hi @Hadatko. I have tried to compile a simple test, but i'm stuck with the linking.
I have created a Makefile which basically runs the two commands bellow. The first builds ok, but the second fails. I'm not sure if it is because of mixing C/C++.

Any suggestion if i have missed something?

First i run "make -f Makefile-eRPC" (which is a copy of /erpc-1.5.0/erpc_c/Makefile)
This will create the lib file, "Debug/Linux/erpc/lib/liberpc.a" under my lokal directory.

Then i run my own Makefile which will generate the following g++ commands:
```
$ g++  -std=c++11 -lpthread -I./inc -I./idl -I./erpc_c/config -I./erpc_c/infra -I./erpc_c/port -I./erpc_c/setup -I./erpc_c/transports  -c -o hello_serverCPP.o hello_serverCPP.cpp
$ g++  -std=c++11 -lpthread -I./inc -I./idl -I./erpc_c/config -I./erpc_c/infra -I./erpc_c/port -I./erpc_c/setup -I./erpc_c/transports -L./Debug/Linux/erpc/lib -lerpc -static  hello_serverCPP.o -o helloServer
```

Output:
```
$ make
g++  -std=c++11 -lpthread -I./inc -I./idl -I./erpc_c/config -I./erpc_c/infra -I./erpc_c/port -I./erpc_c/setup -I./erpc_c/transports  -c -o hello_serverCPP.o hello_serverCPP.cpp
g++  -std=c++11 -lpthread -I./inc -I./idl -I./erpc_c/config -I./erpc_c/infra -I./erpc_c/port -I./erpc_c/setup -I./erpc_c/transports -L./Debug/Linux/erpc/lib -lerpc -static  hello_serverCPP.o -o helloServer
hello_serverCPP.o: In function `__static_initialization_and_destruction_0(int, int)':
hello_serverCPP.cpp:(.text+0x6f): undefined reference to `erpc_generated_crc'
hello_serverCPP.cpp:(.text+0x7b): undefined reference to `erpc::Crc16::Crc16(unsigned int)'
hello_serverCPP.cpp:(.text+0x8a): undefined reference to `erpc::Crc16::~Crc16()'
collect2: error: ld returned 1 exit status
Makefile:133: recipe for target 'helloServer' failed
make: *** [helloServer] Error 1
```

The code is very simple for now.
hello_server.cpp:
```
#include <iostream>
#include <thread>

/* eRPC */
#include "crc16.h"
#include "basic_codec.h"
#include "simple_server.h"
#include "tcp_transport.h"

/* IDL */
#include "hello_world.h"
#include "hello_world_server.h"

using namespace std;
using namespace erpc;

extern const uint32_t erpc_generated_crc;
Crc16 g_crc16(erpc_generated_crc);                     // <--- Fails here

int main(int argc, const char *argv[]){
    cout << "Starting ERPC server...\n" << endl;
    return 0;
}
```

IDL:
```
enum LEDName { kRed, kGreen, kBlue, kYellow }
interface IO {
    set_led(LEDName whichLed, bool onOrOff) -> void
    get_led(LEDName whichLed) -> bool
}
```

```
$ tree -L 2$ tree -L 2
.
|-- Debug
|   `-- Linux
[... 148 lines cut ...]
|       |-- transports.dox
|       |-- uart_cmsis_transport.cpp
|       `-- uart_cmsis_transport.h
|-- idl
|   |-- erpc_crc16.h
|   |-- hello_world.erpc
|   |-- hello_world.h
|   |-- hello_world_client.cpp
|   |-- hello_world_server.cpp
|   `-- hello_world_server.h
|-- inc
|   `-- hello_server.h
|-- mk
|  |-- common.mk
|  |-- erpc_common.mk
|  |-- flags.mk
|  |-- paths.mk
|  |-- subdirs.mk
|  `-- targets.mk
|-- Makefile
|-- Makefile-eRPC
|-- hello_serverCPP.cpp
|-- hello_serverCPP.h
`-- hello_serverCPP.o
```

Hadatko, 2017-12-06:
Hi @CPPZombie , It looks like you are using older erpcgen app where generating crc wasn't supported. I think you can comment lines where crc is used in your main files for client and server.  In never erpcgen crc variable is generated in server and client cpp shim code. And in prepared version is generating crc variable voluntarily.

CPPZombie, 2017-12-06:
Thanks for the clarification.

closed 2017-12-07 by CPPZombie

drumairali, 2021-09-30:
test_unit_test_common_server.h is missing

Hadatko, 2021-10-02:
Hi @drumairali, it is not missing. It is generated from [unit_test_common.erpc](https://github.com/EmbeddedRPC/erpc/blob/develop/test/common/unit_test_common.erpc). You need run erpcgen on this file.