EmbeddedRPC/erpc #14: eRPC server Multiple IO service

chandiniib, 2018-02-19:
Hi Marek and Dusan

I am trying to add multiple service at the server side like below

erpc_server_init(transport, message_buffer_factory);

// Add the IO service.1
erpc_add_service_to_server( create_IO_service1() );
 printf("CAD service added\r\n");
// Add the IO service. 2
 erpc_add_service_to_server(create_IO_service2());

Is it possible to add ?? I am getting error 9, if try to add second service..

Is 2nd service create one more channel ? or use existing one ?

Thank you very much in advance
Chandini

Hadatko, 2018-02-19:
Normally we are using more than 2 services for applications. If you are using old(master) version you need have enough memory for dynamic allocation. If you are using develop version you need have enough memory for static memory initialization. 

Error 9: kErpcStatus_BufferOverrun
Looks like in second service you are calling function which istrying seralize more data than data buffer length is presented. You can try comment first service and try only second one. I think you need increase heap size or buffer length.

chandiniib, 2018-02-19:
Dusan

Thank you for response Dusan Good to know these things ,It is working fine now . We were missing interface id in our case  

Thank you
Chandini

Hadatko, 2018-02-19:
You are welcome. Of course in new version of eRPC you can also use dynamic allocation. Allocation i mentioned is for service allocations. Not for buffer allocations. Buffers depends on choosen message buffer factory class.

closed 2018-02-19 by Hadatko