EmbeddedRPC/erpc #15: .template file editor

nickich, 2018-02-21:
What tools you use to edit/generate the .template files (in "erpc/erpcgen/src/templates/")?

Hadatko, 2018-02-21:
Hi,
*.template files are not generated. They are used to generate *.c, *.cpp, *.py files. For generating files is used txt_to_c.py script.

nchokoev, 2018-02-21:
@Hadatko 
I'll appreciate if you can send me the Notepad++ UDL. I'm try to add a language support and editing those files in plain notepad is a bit of a challenge. 
Thank you.

Hadatko, 2018-02-21:
Files with number 2 i created now for default notepad style. Unlikely i lost my most up to date styles, but these should be pretty up-to-date. I hope these will improve your eRPC apps developing.

[Desktop.zip](https://github.com/EmbeddedRPC/erpc/files/1744433/Desktop.zip)

nchokoev, 2018-02-21:
@Hadatko 
Awesome! Thanks a lot!

closed 2018-02-21 by nchokoev

Hadatko, 2018-02-21:
You are welcome ;)