EmbeddedRPC/erpc #16: Server notification message

nchokoev, 2018-02-22:
How can I define a method, which is called from the server? I guess the "kNotificationMessage" is allocated for that reason... Maybe on the code generator side the "TOK_ASYNC" is the token?

I couldn't find any documentation?
I found about the "kInvocationMessage" with "kReplyMessage"; and about the "kOnewayMessage", but not the notification...

Hadatko, 2018-02-22:
Hi, 
i am sorry for documentation of this area. This is possible but different way.
Here https://github.com/EmbeddedRPC/erpc/wiki/eRPC-infrastructure you can find information about arbitrator. 

If you want communicate bidirectional you need both client and server on both sides and connect them through transport arbitrator and arbitrator client. You need also set task id to server and client task to  differentiate client task form server task (in case you want call RPC function in Server task).

Also you can look closer to this test: https://github.com/EmbeddedRPC/erpc/tree/master/test/test_arbitrator

nchokoev, 2018-02-22:
Hi,
I wanted to avoid having server and client on both sides... What's the reason not using the "kNotificationMessage"? I guess it's thought for that same reason, isn't it?
Thanks.

Hadatko, 2018-02-22:
Hi,
well kNotificationMessage shouldn't be used for eRPC calls from server to client side. If you want call eRPC calls from server to client side, you should use client on server side. I think kNotificationMessage should be used for some certain messages from server side like server failure, ... but it is not implemented currently.

nchokoev, 2018-02-22:
Hi,
Thanks for the explanation. It's clear now :)

Hadatko, 2018-02-22:
Currently tested way is have RTOS on both side with two task on both side. One for client and one for Server (if you will use client eRPC calls (core2) only inside server (core2) implementation of eRPC call from client side (core1) then maybe one task is enough). Didn't tested with Baremetal. They can maybe coexist in Baremetal also in one task or use poll function for running server and after then calling client calls.

closed 2018-02-23 by Hadatko