EmbeddedRPC/erpc #17: Multiple calls between A7 and M4 using eRPC

chandiniib, 2018-03-08:
Hi 

I am trying to understand more about rpmsg here. I would like to run 2 process at the same time between M4 and A7 using eRPC

1.	Lets say Process 'A' is running between M4 and A7 using rpmsg 'A' channel . now I would like call Process 'B' from A7 and that should redirect to proper function in M4 without effecting Process  'A' communication.

2.	Can I achieve this by just creating one more endpoint in A7/M4 or should I need to create one more rpmsg channel as well ?

Thank you so much in advance
Chandini

Hadatko, 2018-03-08:
From my point of view two endpoints are enough. But eRPC can be issue. On one side you can have only one client (in current implementation). If you want more clients you can achieve this by creating client0 and client1 object (instead of client object) And one shim code manualy edit and rename all client to client0 and second from client to client1.

chandiniib, 2018-03-08:
Hi 

That i can do 2 endpoint and 2 client objects. 
Sorry if i am asking wrong question , i am still trying to understand of buffer count importance , currently i am using 8 buffer count with buffer size 5K ..

but if i use 2 end points and 2 client objects then should i need to increase buffer count size ??

Hadatko, 2018-03-08:
Hi Chandini,
What i know: If you have two clients with RPMSG with two endpoints, you need also two servers.
If you are not using oneway function then for this configuration 4buffers should be fine. (one for send one for receive multiply two). If you have using oneway function (mean word ONEWAY in IDL) then it depends on application how much buffers you need.

chandiniib, 2018-03-08:
Hi Dusan

I do use ONEWAY functions and return functions both, but i am just returning true or false for now.

But i use OUT in parameter like ," Function (IN inupt, OUT output) "; on the server side after implementation i do return upto 5k big buffer to client Using OUT parameter not Return type, The question is here if i increase buffer count , do i get faster performance ?

Hadatko, 2018-03-08:
When you are not using oneway functions 4 bufferss are ok. More are optional but will be not used. When you have oneway functions it dependends on application. If you are calling more often oneway eRPC calls then buffers are freed, then your application will fail.

chandiniib, 2018-03-08:
Thank you so much Dusan

closed 2018-03-08 by chandiniib

Hadatko, 2018-03-08:
you are welcome