EmbeddedRPC/erpc #18: using the application as bidirectional

VFSAS, 2018-04-13:
I want to use function calls from M4 and A9 simultaneosly.
I am using arbitrator sources for it.
The function call from A9 works fine and M4 returns data.
But function call from M4 seems doesn't work. The error appears in A9 "Waiting MU transmit buffer empty timeout!
ugh, imx_mu_rpmsg_send() failed: -5". 

After this error data function call doesn't work in another side too: "rpmsg_multiept rpmsg0: virtqueue_add_outbuf failed: -5"

M4 code

.
.

erpc_transport_t transport = erpc_transport_rpmsg_lite_rtos_remote_init(30, 1024, BOARD_SHARED_MEMORY_BASE, 0, NULL, cp);
		    erpc_mbf_t message_buffer_factory = erpc_mbf_rpmsg_init(transport);
		    erpc_transport_t arbitrator = erpc_arbitrated_client_init(transport, message_buffer_factory);
                    erpc_server_init(arbitrator, message_buffer_factory);
		    erpc_add_service_to_server(create_XXX_service());
		  

while (true)
   	  	{
	   erpc_server_poll();
           function1(....);
}

A9 code

.
.

erpc_transport_t transport = erpc_transport_rpmsg_linux_init(1024, 0, 30);
    erpc_mbf_t message_buffer_factory = erpc_mbf_dynamic_init();
    erpc_transport_t arbitrator = erpc_arbitrated_client_init(transport, message_buffer_factory);
    erpc_server_init(arbitrator, message_buffer_factory);
    erpc_add_service_to_server(create_XXX_service());

while (true)
   	  	{
	   erpc_server_poll();
           function2(....);
}
Could anybody suggest anything?

Hadatko, 2018-04-13:
Hi, sorry i couldn't go back to you earlier. Can you post also your erpc_config.h file?

VFSAS, 2018-04-13:
M4
#ifndef _ERPC_CONFIG_H_
#define _ERPC_CONFIG_H_

/*!
 * @addtogroup config
 * @{
 * @file
 */

////////////////////////////////////////////////////////////////////////////////
// Declarations
////////////////////////////////////////////////////////////////////////////////

//! @name Threading model options
//@{
#define ERPC_THREADS_NONE (0)     //!< No threads.
#define ERPC_THREADS_PTHREADS (1) //!< POSIX pthreads.
#define ERPC_THREADS_FREERTOS (2) //!< FreeRTOS.

#define ERPC_NOEXCEPT_DISABLED (0) //!< Disabling noexcept feature.
#define ERPC_NOEXCEPT_ENABLED (1)  //!<  Enabling noexcept feature.

#define ERPC_NESTED_CALLS_DISABLED (0) //!< No nested calls support.
#define ERPC_NESTED_CALLS_ENABLED (1)  //!< Nested calls support.

#define ERPC_NESTED_CALLS_DETECTION_DISABLED (0) //!< Nested calls detection disabled.
#define ERPC_NESTED_CALLS_DETECTION_ENABLED (1)  //!< Nested calls detection enabled.

#define ERPC_MESSAGE_LOGGING_DISABLED (0) //!< Trace functions disabled.
#define ERPC_MESSAGE_LOGGING_ENABLED (1)  //!< Trace functions enabled.
//@}

//! @name Configuration options
//@{

//! @def ERPC_THREADS
//!
//! @brief Select threading model.
//!
//! Set to one of the @c ERPC_THREADS_x macros to specify the threading model used by eRPC.
//!
//! Leave commented out to attempt to auto-detect. Auto-detection works well for pthreads.
//! FreeRTOS can be detected when building with compilers that support __has_include().
//! Otherwise, the default is no threading.
#define ERPC_THREADS (ERPC_THREADS_FREERTOS)//vf

//! @def ERPC_DEFAULT_BUFFER_SIZE
//!
//! Uncomment to change the size of buffers allocated by one of MessageBufferFactory.
//! (@ref client_setup and @ref server_setup). The default size is set to 256.
//! For RPMsg transport layer, ERPC_DEFAULT_BUFFER_SIZE must be 2^n - 16.
#define ERPC_DEFAULT_BUFFER_SIZE (256)//vf

//! @def ERPC_DEFAULT_BUFFERS_COUNT
//!
//! Uncomment to change the count of buffers allocated by one of staticly allocated messages.
//! Default value is set to 2.
//#define ERPC_DEFAULT_BUFFERS_COUNT (2)

//! @def ERPC_NOEXCEPT
//!
//! @brief Disable/enable noexcept support.
//!
//! Uncomment for using noexcept feature.
//#define ERPC_NOEXCEPT (ERPC_NOEXCEPT_ENABLED)

//! @def ERPC_NESTED_CALLS
//!
//! Default set to ERPC_NESTED_CALLS_DISABLED. Uncomment when callbacks, or other eRPC
//! functions are called from server implementation of another eRPC call. Nested functions
//! need to be marked as @nested in IDL.
//#define ERPC_NESTED_CALLS (ERPC_NESTED_CALLS_ENABLED)

//! @def ERPC_NESTED_CALLS_DETECTION
//!
//! Default set to ERPC_NESTED_CALLS_DETECTION_ENABLED when NDEBUG macro is presented.
//! This serve for locating nested calls in code. Nested calls are calls where inside eRPC function
//! on server side is called another eRPC function (like callbacks). Code need be a bit changed
//! to support nested calls. See ERPC_NESTED_CALLS macro.
//#define ERPC_NESTED_CALLS_DETECTION (ERPC_NESTED_CALLS_DETECTION_DISABLED)

//! @def ERPC_MESSAGE_LOGGING
//!
//! Enable eRPC message logging code through the eRPC. Take look into "message_logging.h". Can be used for base printing messages,
//! or sending data to another system for data analysis. Default set to ERPC_MESSAGE_LOGGING_DISABLED.
//!
//! Uncomment for using logging feature.
//#define ERPC_MESSAGE_LOGGING (ERPC_MESSAGE_LOGGING_ENABLED)

//@}

/*! @} */
#endif // _ERPC_CONFIG_H_
////////////////////////////////////////////////////////////////////////////////
// EOF
////////////////////////////////////////////////////////////////////////////////

VFSAS, 2018-04-13:
A9

#ifndef _ERPC_CONFIG_H_
#define _ERPC_CONFIG_H_

/*!
 * @addtogroup config
 * @{
 * @file
 */

////////////////////////////////////////////////////////////////////////////////
// Declarations
////////////////////////////////////////////////////////////////////////////////

//! @name Threading model options
//@{
#define ERPC_THREADS_NONE (0)     //!< No threads.
#define ERPC_THREADS_PTHREADS (1) //!< POSIX pthreads.
#define ERPC_THREADS_FREERTOS (2) //!< FreeRTOS.

#define ERPC_NOEXCEPT_DISABLED (0) //!< Disabling noexcept feature.
#define ERPC_NOEXCEPT_ENABLED (1)  //!<  Enabling noexcept feature.

#define ERPC_NESTED_CALLS_DISABLED (0) //!< No nested calls support.
#define ERPC_NESTED_CALLS_ENABLED (1)  //!< Nested calls support.

#define ERPC_NESTED_CALLS_DETECTION_DISABLED (0) //!< Nested calls detection disabled.
#define ERPC_NESTED_CALLS_DETECTION_ENABLED (1)  //!< Nested calls detection enabled.

#define ERPC_MESSAGE_LOGGING_DISABLED (0) //!< Trace functions disabled.
#define ERPC_MESSAGE_LOGGING_ENABLED (1)  //!< Trace functions enabled.
//@}

...................................................
#endif // _ERPC_CONFIG_H_
////////////////////////////////////////////////////////////////////////////////
// EOF
////////////////////////////////////////////////////////////////////////////////

Hadatko, 2018-04-16:
Hi @VFSAS ,
i assume that in both config you should set 
#define ERPC_DEFAULT_BUFFERS_COUNT (4) //min 4 for bidirectional communication
For sure define in both also: 
#define ERPC_DEFAULT_BUFFER_SIZE (256)
I don't know how old your rpms-lite is, because this condition must be true in newer 
//! For RPMsg transport layer, ERPC_DEFAULT_BUFFER_SIZE must be 2^n - 16.
In older there was issue when size had to be 2^n

VFSAS, 2018-04-23:
Hi, Dusan.
sorry for delay.
It works.
Now I should try to increase the transfer speed.

closed 2018-04-25 by VFSAS