EmbeddedRPC/erpc #19: Data transfer More than buffer size

chandiniib, 2018-04-24:
Hi 
I am setting buffer size for 1k and transferring 1k array data between A7 and M4 without any issue . 
But now i am trying to transfer 2k array blob between A7 and M4 without changing rpmsg buffer size.

Is it possible by using oneway function ?

Thank you
Chandini

Hadatko, 2018-04-24:
HI, do you mean you are trying to split array blob into two messages? Oneway function works as other functions. Difference is that client call is not waiting for answer from server. He will send data and then he is continuing in application. You need be carefully with oneway function. You will probably need more message buffers because client can send more requests on server.

But for example you can send one message as oneway and second as void. Then you need 2 buffers for sending (or you don't if client will execute time consuming section, or is really slow against server). So server will receive two messages and client will wait until second one is processed.

chandiniib, 2018-04-24:
Dusan , Is it possible to transfer without split ?

Hadatko, 2018-04-24:
Hi Chandini, 
actually there is missing implementation for what you request. If I understand correctly you want let eRPC to split your message, send it to server and then concatenate messages...

chandiniib, 2018-04-24:
Hi Dusan

Yes , ur right let say my RPMSG_BUFFER_Size is 1024 , i would like to transfer 2048 byte over my func1 from A7 to M4 ..

closed 2018-05-22 by chandiniib