EmbeddedRPC/erpc #20: Increasing transfer speed between cores

VFSAS, 2018-04-25:
I am use some methodes for measuring application time and interaction time between several devices in the network. I found out that A9 core interactes quite fast for my task, but sending data to and from M4 core makes too much delay for me (2-4 ms and even more). My task is just send some 0/1 flags from A9 to M4 and back. I use functions with only 1 IN binary arrays for it. Using oneway functions makes even more delay between cores(200ms!)
I am triyng to improve my appl and increase my intercore speed.
Which functions/data types etc. should I use optimally?
Any config adjustments in rpmsg protocol or eRPC configs?

Hadatko, 2018-04-25:
HI,
it doesn't make sense that oneway function takes more time than other type of function. Oneway works same way but client is not waiting for request. 

If you have clients and servers on both sides. You can send only oneway functions, and don't wait for responds. Then other side client can send result to first side when server will finish request. 

In rpmsg there is nocopy implementation which we are using in eRPC. RPMSG is keeping buffer until eRPC will call dispose method. I am not sure how it is with Linux but i think it should work. Is your project using this file? https://github.com/EmbeddedRPC/erpc/blob/master/erpc_c/setup/erpc_setup_mbf_rpmsg.cpp

VFSAS, 2018-04-25:
Hi, Dusan.
Yes, I use this file for M4 application and erpc_setup_mbf_dynamic.cpp for A9.

Hadatko, 2018-04-25:
Other improvement can be: 
Bid data allocate in shared memory. Then you can send only pointer to data through eRPC.
IDL:
 oneway sendA(int[1000] a @shared)
App:
int *a = shared_alloc(1000)
sendA(a);

-> then eRPC message buffer can be smaller.

VFSAS, 2018-04-25:
Matrix multiply example in erpc-imx-demos contains pointers. But when I generate sources using erpcgen there are not pointers, so I use it as it is. Should I use pointers?

Hadatko, 2018-04-25:
Try change this line: 
erpcMatrixMultiply(in Matrix matrix1, in Matrix matrix2, out Matrix result_matrix) -> void
to this:
erpcMatrixMultiply(in Matrix matrix1, in Matrix matrix2 @shared , out Matrix result_matrix) -> void

If there is no change, you will need newer eRPC. The best is most up to day commit on develop branch on github.

VFSAS, 2018-04-28:
Sorry, Dusan,
can't understand shared_alloc/alloca_shared function. Is it form eRPC library?

Hadatko, 2018-05-02:
Hi VFSAS,
sorry, i will try explain. When you use @shared annotation in IDL, data are not serialized. Only pointer to data is sent. For that reason the data need be created in shared memory. And currently they need have same address mapping from both sides. 

For example if **struct A** will be allocated on shared memory address **0x00FF3265** then this address need be accessible and same for both **A-side and M-side**. Allocation **struct A** in shared memory is currently on you (you can take one rpmsg buffer in you application outside of eRPC call and allocated data inside,...).

ablaylock, 2018-05-31:
One additional issue I have found with the performance relates to the performance of the rpmsg implementation. First there is a bunch of debug code in the virtio_rpmsg_bus.c file for linux that prints every message sent to dmesg and there was an issue with rpmsg_xmit_done not releasing a queue so we would wait for upto 15 seconds for the buffer to be returned. The transfer speeds went from less than 1MB/s to ~90MB/s and the message round trip times were greatly reduced.

VFSAS, 2018-06-06:
I don't want to close this issue because I haven't got needed perfomance.
What I've tried:
1. Using FreeRTOS or baremetal on M4,
2. M4 eRPC server - A9 eRPC client or M4 eRPC client/server - A9 eRPC client/server,
3. Using inout structure and sending pointer from A9 to M4,
BUT data transfer delta latency can be up to 16,3 ms!!! to send data from M4 to A9!!! I used oscillograph.
Unfortunately, it's not yet obvious for me how I can use @shared annotation. I have sent pointer from A9 to M4 and pointer addresses were the same, but the dereference values in this case were not 1 or 0 as I want and got in p.3.

chandiniib, 2018-06-12:
Hi Dusan

I am interested to use @shared , i am using latest eRPC from develop and rpmsg-lite from master .

**IDL:**
program erpc_performance
type Number1KByte = uint8[1024];
struct A{Number1KByte number1 }
interface PerformanceService{
	erpcSend1kByteRCV1kByte(A var @shared ,out Number1KByte result)->void
}

**erpc_performance..h:**
void erpcSend1kByteRCV1kByte(const A * var, Number1KByte result);

I could see **On client side** codec->writePtr and **On server**: codec->readPtr

**A7 side App:**
Number1KByte number1kByte,result_number1kByte;

std::allocator<uint8_t> alloc;
auto a = (std::allocate_shared<uint8_t> (alloc,sizeof(A)));
erpcSend1kByteRCV1kByte(a,result_number1kByte);

But after client init everything hangs,  Could you please tell me How to use @shared ?

Hadatko, 2018-06-12:
Hi Chandini,
i don't have much experience with Linux. I would suggest get one rpmsg buffer in application (which will serve for storing shared data not to send buffer to other side). Copy your data of structure A into it. Try send address of structure A from this buffer.
If your application is hanging in client init before calling eRPC function then issue is somewhere in init function....

chandiniib, 2018-06-12:
Hi Dusan
You are right , it hangs some where in client_init . i have removed @shared from idl and thought to run simple example but it hangs in the same place

VFSAS, 2018-06-13:
@ablaylock 
Have you finally achieved appropriate speed between cores?
Because I've tried to work without eRPC and I faced too with astonishing delay you mentioned.

ablaylock, 2018-06-13:
Unfortunately I do not have a kernel patch at this time.

To patch the module remove all `prink_hex_dump` calls in `virtio_rpmsg_bus.c` and any adjacent `dev_dbg` calls. This may create issues with the M4 side depending on the RTOS or layer you are using. I was using CMSIS-RTOS (KEIL RTX) and had issues with the interrupt message queue overflowing due to the increase in message speed.

pazzodalegare, 2018-06-13:
Thank you for your fast answer.
I've just removed prink_hex_dump and I got some improvements (from 90KB/s  to around 140KB/s). 
What about the issue on rpmsg_xmit_done, could you please give more info/patch?
Thanks!
pz

ablaylock, 2018-06-13:
There is probably a better solution than this for the `rpmsg_xmit_done `but I think what I did was change how new buffers are acquired in `rpmsg_send_offchannel_raw`. There is a loop that calls wait_even_interruptible_timeout and it waits 15 seconds before timing out but for whatever reason it seemed like under certain conditions it would wait 15 seconds despite there being available buffers. I reduced the time from 15s to something very small  (like us time scale) and then had the timeout be a result of the number of calls to get a tx_buff that failed.

VFSAS, 2018-06-14:
@ablaylock @Hadatko @pazzodalegare @chandiniib 
Dear colleagues, this is very similar to issue that  @ablaylock mentioned.
https://github.com/EmbeddedRPC/erpc-imx-demos/issues/3

chandiniib, 2018-06-14:
Thank you @VFSAS

pazzodalegare, 2018-06-15:
@ablaylock, @VFSAS : thank you

VFSAS, 2018-07-17:
@Hadatko @ablaylock @pazzodalegare @chandiniib 
Finally, I've got guite good performance with eRPC. I was playing with linux and also source files as @ablaylock suggested.
I think, we'll implement eRPC in our new devices. Good work, Dusan.

closed 2018-07-17 by VFSAS

MarekNovakNXP, 2018-10-24:
In the new kernel (4.14), there is a config macro: CONFIG_DYNAMIC_DEBUG.
If you disable this macro, you should get the speed improvement.

sharmiladevadas, 2019-01-11:
Hi,
I am using RPMsg framework to send data between Cortex-M4 to Cortex-A35 core. Can you please tell me how to find the throughput of the data transfer using RPMsg.
Thank you