EmbeddedRPC/erpc #21: Annotation 'include'

nchokoev, 2018-05-18:
Hi,
What's the right use of '@include("myheader.h")' annotation? It doesn't seem to work for me.

Hadatko, 2018-05-24:
Hi,
When this annotation is used, you should see this include in generated common header file. This header file can contain for example data types and functions which are defined as @external in IDL. For which version of eRPC it is not working for you? This annotation need be assigned to "program" keyword.

Example:
IDL:
@include("foo.h")
program my_program
 ...

Example of usage can be found in erpc/test/test_annotation/test_annotation.erpc

nchokoev, 2018-05-30:
Hi,
It works for the 'program' keyword.
I left with the impression that it should work for 'interface' as well, looking at the doc
https://github.com/EmbeddedRPC/erpc/wiki/IDL-Reference
Is there a way to specify includes to be added not in the common header file, but in the server or client header files separately...

Hadatko, 2018-06-03:
Hi, if you are using actuall master branch, then documentation can be wrong. We are putting on first page of repository commit sha for correct documentation on wiki. I didn't realized that you need clone wiki repo to checkout to correct branch. We will need work on this synchronization better.

closed 2018-09-27 by Hadatko