EmbeddedRPC/erpc #23: What's the best way to transfer a file, or any non fixed size of binary data, with eRPC?

Masterbar, 2018-05-31:
Hi,

We use eRPC between an ARM/QNX and a Kinetis/FreeRTOS (connected with SPI) for some remote functions calls.
Now we would like to transfer a new FW image to the Kinetis MCU, if possible using the eRPC server running on the Kinetis. What's the best way to do that? Which type should we use in the IDL?
Notice that we may also need to transfer other files (e.g. XML config file).

Thank you for your help,
Greg

Hadatko, 2018-05-31:
Hi @Masterbar,
if non-fixed size means that size can be changed, but you always now before sending current size, you can use list/binary types. For example:

list<uint8> will create in C
```
struct list
{
uint8_t data; //pointer to data
uint32_t size; // size of data
}
```

if you don't like structure you can use annotation to specify size
`myF(list<uint8>data @length(size), uint32 size)` -> will not create structure.

Hadatko, 2018-05-31:
Binary format can be faster and better in this case. Binary is like list<uint8> but list is copying data in for loop while binary is copying using memcopy.

Masterbar, 2018-06-04:
Hi @Hadatko ,

How could we avoid dynamic allocation of memory (server side) when using the binary type?
Now I have in my IDL something like:
`sbc_write(binary data) -> int32`
and this generates in my server:
`data->data = (uint8_t *) erpc_malloc(data->dataLength * sizeof(uint8_t));`
As we are running the server on a MCU, we would like to use only static memory.

Notice that we already use erpc_mbf_static_init().

Best Regards,
Greg

Hadatko, 2018-06-04:
Hi @Masterbar,
this is one of our important task into future. Now it can be done by replacing implementation of erpc_malloc function. But we don't have this implementation. You need do it right now on your own.

closed 2018-09-27 by Hadatko