EmbeddedRPC/erpc #24: Implementing threading support for CMSIS-RTOS2

ablaylock, 2018-05-31:
I would like to implement threading support for CMSIS-RTOS2 and have ported most of the methods needed. I am a little unsure of the intended use of Thread::threadEntryPoint, and Thread::threadEntryPointStub. It looks like threadEntryPointStub is used to call the actual threadEntryPoint on the casted class passed by arg. One particularly confusion point in the freertos port is the Thread* linked list and its purpose. It appears that the pthread example doesn't have this linked list and neither does the zepher implementation. My best guess for the purpose behind the linked list is for the getCurrentThread function. Is this correct?

Hadatko, 2018-06-03:
Hi ablaylock, i quickly look on implementation and it looks you are right.  Zephyr and pthread don't need linked list. Only freertos for reason you described.

Threadentrypointstub is used to call threadEntryPoint and also for cleaning stuff after entry finish (like you see in freertos port).

ablaylock, 2018-06-04:
Thank you for the information.

closed 2018-06-04 by ablaylock