EmbeddedRPC/erpc #28: How to convert C header to eRPC IDL file

sudhakarreddi, 2018-06-07:
Hi All,

I would like to use eRPC for one of my project. I am looking for option, is there any way to convert C language header file to eRPC IDL file. I have a large chunk of C header file so I am finding it difficult to write the eRPC IDL file manually.  Appreciate your help in this regard. Thank you for your help.

Regards,
Sudhakar

Hadatko, 2018-06-07:
Hi @sudhakarreddi ,
this is also one of our goal. To create parser for C header file. But it is not done yet.

sudhakarreddi, 2018-06-08:
Hi @Hadatko ,
Thank you for quick response.  Is it coming any time sooner ? If yes I probably wait for some time. Please let me know ?

Regards,
Sudhakar

Hadatko, 2018-06-12:
Hi, 
currently we don't have estimation for new features.

nchokoev, 2018-09-03:
We use python script and ctags (http://ctags.io/) to convert c headers to .erpc.

closed 2018-09-27 by Hadatko

Pratyush-Mallick, 2021-12-01:
> We use python script and ctags (http://ctags.io/) to convert c headers to .erpc.

Hi Nikolay, 

Can you provide more information on this?

pmallick-adi, 2021-12-06:
Someone please help me with this.