EmbeddedRPC/erpc #29: '@group()' annotation

nchokoev, 2018-06-12:
Hi,
When using '@group()' for an interface it produces the correct files, except the data types are duplicated in all the headers (for all groups). 
The interface related functions are OK in the source files as well as in the headers. The interface IDs and function IDs are fine (per group). All other C types are duplicated in the header files for each group.
Looking at the templates it doesn't seem to handle types separation per group...

Hadatko, 2018-06-12:
Hi that should be ok this duplication because they are always same. It is because several groups can use same data types. If you don't like it, you can define for program @types_header(string)  where all data type declarations will be stored in one file. More in documentation.

nchokoev, 2018-06-12:
Awesome. Thanks.

closed 2018-06-12 by nchokoev