EmbeddedRPC/erpc #39: TCP Example client / server code

sudhakarreddi, 2018-06-29:
Hi All,

I am new to eRPC. Is there any example of client & server initialization code in C language, with erpc using TCP as transport mechanism ?

Can you please point me to the code, if it is readily available.  I thank you for your support.

Regards,
Sudhakar

Hadatko, 2018-07-03:
Hi guys, these days i don't have much time (Vacations, others projects, wedding ;) ... ). So what i can suggest now is look on implementation of other transports in erpc/transports and erpc/setup folders. Tcp transport is currently only for using on Linux (this transport doesn't have C setup function in erpc/setup folder currently, but can be easily done when you look on other transports).
 When you look into transports folder you will see that you basically need implement 2-3 functions (send, receive, init in C++ but simple). Then C implementation of calling this C++ functions are placed in setup folder. I don't see big issue to implement this with lwip or other tcp/ip stacks or other transports.
In init function you can implement socket initialization and in send/receive you will call lwip send/receive functions.

kschuemann, 2018-07-04:
Hi Hadatko, 

thanks for the hints. I will give it a try and give you feedback during process. Best wishes for the wedding. 

Regards,
Kolja

mentaal, 2018-07-04:
I do have this pull request in place for a small piece of TCP initialization code if it helps:
https://github.com/EmbeddedRPC/erpc/pull/35

closed 2022-05-03 by Hadatko