EmbeddedRPC/erpc #41: Code generator crash

nchokoev, 2018-09-03:
The code generator crashes when an enum member is initialized with negative value. Simple .erpc file:
`program enumtest
enum error_e
{
	ERROR_NONE = 0,
    ERROR_UNKNOWN = -1
}

interface etest
{
    enum_test(error_e e) -> int32
}`

The 'assert(member->hasValue());' asserts...

Hadatko, 2018-09-03:
Hi, thank you for your comment. Meantime this will be fixed you can use value -2.

nchokoev, 2018-09-03:
Hi Dusan,

The value of -2 causes the same exception when the member is initialized as
'-2'.
I'm not sure what's the initial intend of the 'hasValue()' logic. I'm thing
that the assumption is that when the 'm_value.getValue()' returns something
other than '-1' the member is initialized...
Looking at the code seems that the enum members are always initialised
within eRPC, so I 'fixed' it for now by simply always returning 'true' for
'hasValue()'. The side effect is that on the result code all enum members
are being assigned, which is not that bad.
I was thinking also to remove the cast to 32 bit, but it will fail for 64
bit negative values then (now it may fail as well for 64 bit ints...)
Maybe additional member to specify if the member is initialized is needed
in the 'IntegerValue' class. It can be initialised from:
'AstNode *SymbolScanner::handleEnumMember(AstNode *node, bottom_up)'
in the 'enumMemberHasValue(node)' case, where the behavior seems to be
correct...

Regards,
Nikolay

dpfrey, 2021-01-22:
This bug is affecting my use of eRPC as well.  I have attempted to fix this in #159.

MichalPrincNXP, 2021-02-11:
Fixed within #159. Thank you.

closed 2021-02-11 by MichalPrincNXP