EmbeddedRPC/erpc #42: Issue with comments

nchokoev, 2018-09-05:
Hi,
I'm facing an issue with the comments within an enum
1. C-style doxygen comments within an enum are not present in the generated code. No error message.
In the .erpc file:
```
enum test_enum
 {
	/* 0 */
    TEST_0 = 0,
	/*! 1 */	
    TEST_1 = 1,
	/*! 2 */
    TEST_2 = 2,
	/*! 3 */
    TEST_3 = 3
	/* Bottom comment */
}
```

Generates:
```
typedef enum test_enum
{
    TEST_0 = 0,
    /*! 1 */
    TEST_1 = 1,
    /*! 2 */
    TEST_2 = 2,
    /*! 3 */
    TEST_3 = 3
} test_enum;
```
(First and last comments are missing)

2. A doxygen comment following the enum members declarations causes an error:
In the .erpc file:
```
enum test_enum
 {
	/* 0 */
    TEST_0 = 0,
	/*! 1 */	
    TEST_1 = 1,
	/*! 2 */
    TEST_2 = 2,
	/*! 3 */
    TEST_3 = 3
	/*! Bottom comment */
}
```

causes:
"error: file test_comments.erpc:14:1: syntax error, unexpected '}', expecting identifier"

Hadatko, 2018-09-05:
First missing comment is not doxygen comment. So it will not appear in generated output. Non doxygen comments are currently only for IDL and will not be transformed in generated output. Where doxygen comments can be put was designed from https://www.stack.nl/~dimitri/doxygen/manual/docblocks.html
Try to use as a last comment TEST_3 = 3 /*!< Bottom comment */

nchokoev, 2018-09-06:
So, it doesn't support regular comments, but only Doxygen. ...and forces Doxygen rules.
Thanks for the explanation.

closed 2018-09-06 by nchokoev