EmbeddedRPC/erpc #43: Serial test fail

jpacheco8, 2018-09-11:
When I run any serial test in Linux the following error appears:

Starting ERPC server...
test_lists_server_serial_test: /erpc-master/erpc_c/infra/erpc_framed_transport.cpp:67: virtual erpc_status_t erpc::FramedTransport::receive(erpc::MessageBuffer*): Assertion `m_crcImpl && "Uninitialized Crc16 object."' failed.
Aborted

closed 2018-09-11 by jpacheco8

LiePy, 2021-12-08:
I also couldn't successfully test serial port based eRPC, I used virtual serial port created by "socat", maybe it doesn't support virtual serial port. I want to know if there is any problem with my code. 
    Do we have the eRPC demo based on serial port？

MichalPrincNXP, 2021-12-14:
Hi, unfortunately there is no example showing the serial transport usage. It deals with not-maintained transport.