EmbeddedRPC/erpc #47: Callback on client side

MarkusHolzl, 2018-10-09:
Hello eRPC-Team,
thanks for your project, great work! I’ve a question regarding the callback feature.
Is it possible to register a callback function reside on the client to the server? Is it possible to call the callback function on the client from the server? Could you provide an IDL example for this?
Best regards, MH.

Hadatko, 2018-10-09:
And thank you for your feedback ;) 

Allthough i just remembered that we have callback example directly on github. You need browse test/test_callback folder. There is some simple eRPC test for callback support. You can run test under Linux. There is some GoogleTestFramework balast. But if you saw client and server application it should make sense to you.

MarkusHolzl, 2018-10-09:
Thanks for your quick answer.
Yes, we've seen the examples, but:
a.) in the example case, the client register a callback function implemented and called on the server side,
b.) in our case the server should call(back) a function implemented on the client. The client should register his callback function to server before.
Is case b.) possible with eRPC?
Because in case b.) the server has to call a (callback) function implemented on the client. I'm not sure if this case is provided by eRPC.

Hadatko, 2018-10-09:
Callbacks are a bit complicated in compare with other cases. I will start with simple YES it is possible ;) callback function can be always be called on client/server side. It doesn't matter where is original definition and where is eRPC definition (one is on cliend side and one on server). I think all cases are covered in test application i mentioned.
https://github.com/EmbeddedRPC/erpc/blob/master/test/test_callbacks/test_callbacks.erpc
In IDL you can see callback functions on client side and on server side. 
Client is sending callback to server and server can call this callback, or it can return this callback back and client can call this callback.  
Client is receiving callback from server (out parameter) and he can call this callback. 
You need a bit more investigate this. I can share with you picture how it is working but it is old design but similar.
[image]

If you will have more question, i will try answer my best.

jaggi, 2019-06-10:
Hi Hadakto,
I am bit confused with the implementation of callbacks in the current example of test_callbacks .
Can you please share me the flow of implementation of callbacks for the current example of callbacks,so that it will be helpful for me.

Thanks in advance...!

MichalPrincNXP, 2019-06-10:
Hello @jaggi,
the flow of the test_callbacks is as follows:
The test named _In_Out_table_:
1. The client side calls the myFun() with two params. The first parameter is the callback function name that should be executed on the server side. The second parameter is the pointer on the callback1_t type object which should be filled by the server side (out param).
2. Consequently, the myFun() is executed on the server side. It is possible to call the callback function, the callback1() could be called in this test case. The implementation of the callback1() is void, the function body is empty - this should be probably changed a little to ensure the service function really executes the requested callback function. 
3. The pointer to the callback1() function is returned via the second parameter of the myFun() service function. 
4. This return value is then compared on the client side.

The test named _In_Out_withoutTable_:
1. The client side calls the myFun2() with two params. The first parameter is the callback function name that should be executed on the server side again. The second parameter is the pointer on the callback2_t type object which should be filled by the server side (out param).
2. Consequently, the myFun2() is executed on the server side. It is possible to call the callback function, the callback3() could be called in this test case. Looking into the erpc file, the callback3 belongs to the ClientCore1Services interface, i.e. it should be implemented and executed on the client/core0 side. This would require to modify the test framework and to setup client and server on both core0 and core1 sides. To simplify that, the callback3() is implemented on the server/core1 side instead, see the comment /* will be shim code in real use case */ 
3. The pointer to the callback3() function is returned via the second parameter of the myFun2() service function. 
4. This return value is then compared on the client side.

In general, the test_callbacks logic should be modified a little to better understand the flow and to really trigger requested callback execution. I will try to prepare some test update on that.
Also, note that the new erpc example called erpc_two_way_rpc_rpmsg_rtos will be part of the MCUXpressoSDK 2.6.0 that will be released very soon, see the comment from @Hadatko . You can try to download the package for lpcxpresso54114 and have a look at this example that shows how both the eRPC client and the eRPC server can be setup on one side/core (bidirectional communication) and how to handle callback functions in eRPC.

jaggi, 2019-06-12:
Hi MichalPrincNXP,
Thanks for your explanation. It helped me a lot...!

burgerkingorama, 2019-07-01:
Hi guys,

does any one have had success in implementing a client side callback called by server (the green path in the graphic above)?

For me the function pointer argument in `setCallback `is always `NULL` and thus not callable in the server rpc implementation.

MichalPrincNXP, 2019-07-18:
Hello @burgerkingorama,
I recommend to visit the https://mcuxpresso.nxp.com/ and build and download the sdk package for LPCXpresso54114 board. Multicore erpc examples are placed in board/lpcxpresso54114/multicore_examples folder then. Have a look at the erpc_two_way_rpc_rpmsg_rtos example that shows different callback scenarios incl. the one you are referring. 
Here is the updated description of the example:
_The Multicore eRPC Two Way RPC RTOS project is a simple demonstration
program that uses the MCUXpresso SDK software and the Multicore SDK to show
how to implement the Remote Procedure Call between cores of the multicore system.
This multicore example shows how both the eRPC client and the eRPC server can be 
setup on one side/core (bidirectional communication) and how to handle callback
functions in eRPC.  
The primary core (Core0) creates client and server tasks first. The client task
releases the secondary core from the reset, initializes the RPMsg-Lite erpc transport
and once the server task is running it configures the arbitrated client. Then the
application logic is running.
The secondary core (Core1) creates client and server tasks two. The client task 
initializes the RPMsg-Lite erpc transport and once the server task is running it 
configures the arbitrated client. Then the application logic is running.
The client task logic of the Core1 is very simple, it repeatedly calls the
increaseNumber() erpc function that is implemented on the Core0 and that increments
the counter.  
The client task logic of the Core0 alternately issues either getNumberFromCore0() 
function implementation on the Core0 or getNumberFromCore1() function implementation 
on the Core1 (erpc call). Then, the nestedCallGetNumber() erpc function call is issued
that alternately triggers either getNumberFromCore1() function implementation 
on the Core1 (normal erpc call) or it triggers the getNumberFromCore0() function 
implementation on the Core0 (nested erpc call, routed through the Core1 erpc server)._

Regards
Michal

Hadatko, 2019-07-21:
Hi @burgerkingorama , NULL is good enough in some cases. If you have only one callback function. Serialization of function id is skipped beacause is only one. On other side you should see always calling that one function during execution (or assigning that one function). Or try add one more callback to IDL and NULL shoudl dissapear.

jaggi, 2019-08-06:
Hi,
The callback which has been implemented in test_callbacks is of filling pointer on the server side. But can anyone suggest me in such a way that pointer should be filled and called in server side and that callback definition should be on client side.

Thank you...!

MichalPrincNXP, 2019-12-20:
test_callbacks logic improved in [v1.7.3](https://github.com/EmbeddedRPC/erpc/releases/tag/1.7.3) to be more understandable and to allow requested callback execution on the server side.
Improved functions naming and function parameters naming in the test_callbacks to be more understandable.

closed 2019-12-20 by MichalPrincNXP