EmbeddedRPC/erpc #50: self.length error

bataliero, 2018-11-21:
Hi all,
I have small problem, when I run code generated by the command below I get error about invalid object attribute:

`erpcgen/erpcgen -gpy dir/example1.erpc`

Iterface file:
```
program example1

interface Example1
{
    func1(list<uint32> data) -> uint32
}

```

Generated file (client.py):
```

# Generated by erpcgen 1.7.1 on Wed Nov 21 17:40:42 2018.
#
# AUTOGENERATED - DO NOT EDIT
#

import erpc
from . import common, interface

# Client for Example1
class Example1Client(interface.IExample1):
    def __init__(self, manager):
        super(Example1Client, self).__init__()
        self._clientManager = manager

    def func1(self, data):
        # Build remote function invocation message.
        request = self._clientManager.create_request()
        codec = request.codec
        codec.start_write_message(erpc.codec.MessageInfo(
                type=erpc.codec.MessageType.kInvocationMessage,
                service=self.SERVICE_ID,
                request=self.FUNC1_ID,
                sequence=request.sequence))
        if data is None:
            raise ValueError("data is None")
        codec.start_write_list(self.length)
        for _i0 in data:
            codec.write_uint32(_i0)

        # Send request and process reply.
        self._clientManager.perform_request(request)
        _result = codec.read_uint32()
        return _result

```
Problematic line is      
`codec.start_write_list(self.length)`
It seems it should be
`codec.start_write_list(len(data))`
instead.

Am I using IDL  incorrectly or this is a a bug?
Regards

Hadatko, 2018-11-22:
Hi @bataliero,
thank you for this bugfix. We will provide fix in next update. Until then, you can use your fix in local copy.

Hadatko, 2018-11-22:
In https://github.com/EmbeddedRPC/erpc/blob/master/erpcgen/src/templates/c_coders.template
replace **self.length** with **len({$name})**

Hadatko, 2018-11-30:
Hello, i pushed changes to develop branch. But still erpcgen app need be generated on local pc until we will release new app.

closed 2019-10-31 by Hadatko