EmbeddedRPC/erpc #51: Forwarding eRPC failures to application

Mercury2018, 2019-02-11:
Hi all, 
We're using eRPC for communication between 2 applications hosted on 2 different systems for a life safety critical medical product. 
So, we've to handle all the potential error cases at the product level. Being said that, I performed an exercise to capture the error codes from eRPC to the application level. 
I've set a custom error handler to eRPC client to forward the errors to the application. 
On course of this exercise, I found a few observations below, which would need update in a few eRPC files and autogenerated (CPP) files. I request you to kindly have a look and let me know your thoughts. 

1. In autogenerated client file, error from g_client-> performRequest is not handled, rather the error from codec status (err = codec->getStatus();) is returned by error handler. 
I think the codec status has to be checked prior to even performing the request and if codec status is not successful, then the request shall not be performed. 
Also, the error from performRequest has to be handled. 

2. In erpc/infra/src/erpc_transport_arbitrator.cpp, TransportArbitrator::receive(), if there is  error from codec status, it says to continue, It should rather return the error.

3. In TransportArbitrator::prepareClientReceive(), if the PendingClientInfo is NULL, it returns kErpcStatus_Fail which is not complying the method prototype and skips the error from its caller too. 

4. In erpc/transport/src/erpc_tcp_transport.cpp, the TCP_DEBUG_ERR causes the application to exit, by ePRC itself, rather than forwarding it to application.

Request your inputs on the above observations.

Mercury2018, 2019-02-21:
Would like to know, is anyone assigned for this, yet?

Hadatko, 2019-02-21:
HI @Mercury2018 ,
sorry for delay in answer. I hoped i can go trough your comment earlier, but unfortunately i am busy these days. Thank you for this summary. We will take them into account once we will have time.

Mercury2018, 2019-02-21:
Hi,
Thanks for your information.
Would be helpful for my project, if I could get the response as early as possible.

Regards,
Tharini.

mgiaco, 2019-04-11:
Hi @Mercury2018 are you really using erpc for safety. Which target are you using?
many thanks.

Mercury2018, 2019-04-11:
Yes, We're using it for a new product development which is a safety critical medical device. 
We're using it for communicating between 2 applications running in IMX6 Solo X (client) and STM32 H7 (server) respectively.

mgiaco, 2019-04-29:
That sounds great, do you share the STM32 H7 port for erpc?

Mercury2018, 2019-05-07:
Am sorry, your question is not clear to me. Can you please elaborate?

mgiaco, 2019-05-07:
Okay so you write you are using it for IMX6 with an STM32H7.  And I am looking for some code example for the STM32H7. Are you using CMSIS UART or what?

Mercury2018, 2019-05-13:
We're using eRPC over network stack. 
I feel this discussion is deviating from the reported issues. Is there any resolution for the reported issues?
I believe the reported issues can be reproduced with any existing systems supported by eRPC communication.

Hadatko, 2019-05-15:
Hi @Mercury2018 ,
sorry for late answer. To reduce amount of source code we removed error checking from generated code. performRequest should check codec status at beginning. If any error happens he should set error. This should two your bullets. But set error into codec status looks weird to me. Maybe it will be better to change infrastructure a bit and status would go over all infrastructure in some general object.
Continuing when received wrong message means that the message has broken head (beginning of message). But i agree that it should be handled better way most likely.
Last one is for debbuging. I think here it is ok.

Mercury2018, 2019-06-06:
Hi @Hadatko, 
Thanks for your response. 
Does V1.7.1 have removed error checking from generated code? 
And the changes that you're proposing will be part of V1.7.2? Can you please let me know so that it'll be helpful for us to plan accordingly.

MichalPrincNXP, 2019-06-06:
Hello @Mercury2018,
error checking had been removed from the generated code before 1.7.1.
As for the changes proposed by @Hadatko , they are not going to be part of the official v1.7.2, unfortunately. But, we are working on it now and you could expect these bugfixes on develop branch soon (once reviewed and tested properly).

Mercury2018, 2019-06-06:
Hi @MichalPrincNXP, 
Thanks for the update.

MichalPrincNXP, 2019-12-20:
The ClientManager and the ArbitratedClientManager updated in [v1.7.3](https://github.com/EmbeddedRPC/erpc/releases/tag/1.7.3) to avoid performing client requests when the previous serialization phase fails.

closed 2019-12-20 by MichalPrincNXP