EmbeddedRPC/erpc #53: [BUG] SimpleServer destructor deletes static object

eugene345, 2019-03-13:
Hello,

Service objects are statically allocated. But SimpleServer destructor deletes it:

```
SimpleServer::~SimpleServer(void)
{
    while (m_firstService != NULL)
    {
        Service *firstService = m_firstService;
        m_firstService = m_firstService->getNext();
        delete firstService;
    }
}
```

Hadatko, 2019-03-13:
Hi @eugene345 ,
you are correct. Because services are not created inside SimpleServer the solution should be remove freeing services here and provide functions to free them in application where they are created.

MichalPrincNXP, 2019-12-20:
Erpcgen v1.7.3 has been updated to generate the shim code for destroy of statically allocated services, see https://github.com/EmbeddedRPC/erpc/releases/tag/1.7.3

closed 2019-12-20 by MichalPrincNXP