EmbeddedRPC/erpc #54: C examples

mgiaco, 2019-03-20:
So I read: Designed to work well with C, but flexible enough to support object-oriented languages like C++.
Are there any C examples available. 
I would like to use erpc to talk between to STM32 devices. Do you think this is the right lib to use for that.
I think so but I am wonder why I do not found any example for that in the WEB. 

Many thanks
cheers
mathias

Hadatko, 2019-03-20:
Hi @mgiaco ,
this document can help you find C examples for NXP boards: https://community.nxp.com/docs/DOC-334083

mgiaco, 2019-03-20:
Okay many thanks.  Do you mean that. 

Multiprocessor examples
frdmk22f, frdmk28f, frdmk64f, frdmk66f, frdmk82f, frdmkl25z, frdmkl27z, frdmkl43z - MCUXpresso download page

So there is no example on Github - right?

Hadatko, 2019-03-20:
Yes multicore/multiprocessor examples. Better to choose frdm board instead of imx (for simplicity). I don't have knowledge that somebody created C example for any board on github.

mgiaco, 2019-03-20:
Do you think it is complicated to port on other devices? Do you know any port other than NXP device?
thanks

Hadatko, 2019-03-20:
eRPC depends on OS and transport layer.
Os related port files can be found here: https://github.com/EmbeddedRPC/erpc/tree/master/erpc_c/port
Transport related files : https://github.com/EmbeddedRPC/erpc/tree/master/erpc_c/transports and C wrapper https://github.com/EmbeddedRPC/erpc/tree/master/erpc_c/setup

Guide is here: https://github.com/EmbeddedRPC/erpc/wiki/Getting-Started

I don't think it is hard to port. But i am developer of that so i am not fully objective. But Definitly you can get image of whole of that from links above. 

Your case is multiprocessor examples. Freertos? Uart/SPI? Bidirectional communication (requests from both sides) or sending request from one board to second only? I think if you also look into NXP multiprocessor examples for frdm board master/slave examples it can help you understand.

mgiaco, 2019-03-20:
Many thanks, my case: 2 STM32 (sorry for that :-) ) with FreeRtos, GNU ARM GCC. UART or SPI both is available on the board. Bidirectional communication is need but not much data overall.

Hadatko, 2019-03-20:
So at least you don't need to port OS related files. Likely you will need port transport(maybe) which is pretty straightforward (init, send, receive functions). 

Bidirectional in way to send requests from both cores you will likely need two independent communication channels for these transport layers. 
In a way to execute function and get return data it is enough to have one communication channel.
I would suggest to try basic port erpcMatrixMultiply example at first.

mgiaco, 2019-03-20:
many thanks

Hadatko, 2019-03-21:
You are welcome. I am happy to help you.

closed 2019-03-21 by mgiaco