EmbeddedRPC/erpc #56: Timeout for eRPC call

chandiniib, 2019-04-18:
Hi 

Is there any way we can set timeout from client[A7] so that if M4 die's by any case A7 get notification so that we can reset the connection .

Thank you
Chandini

anirudhgargi, 2020-07-02:
@chandiniib Hello Chandini, 

I am using the eRPC and i have the same need. 
Did you  find a way for this requirement of timeout ? Any suggestions for me to handle such scenarios. 

Thanks.

Hadatko, 2020-09-25:
Hi guys,
I did this PR #130. Please take a look into it and comment if it will fit your needs. But this implementation is only for C, but if you are interested in python support you can create own PR inspirated by mine.

pgu-swir, 2020-09-27:
Just as a curiosity, why would a timeout in the transport itself not allow that to be solved? If one core dies in the middle of an RPC, wouldn't the transport on the other one, waiting for a RX or TX, timeout and then action can be taken?

anirudhgargi, 2020-09-27:
Hello @Hadatko, Thanks for the patch. That's a useful addition, adding pre and post handler to call let's you handle error scenarios in better ways. Will see how to implement it.

@pgu-swir: Yes, you are correct. We added a client timeout in transport layer in case something goes wrong on Server side. Helps unblock the execution on client side.

Thanks guys.

Hadatko, 2020-10-03:
Can i close this issue? Or do you think it make sense to keep it open?

Hadatko, 2023-01-04:
Closing due to inactivity and issue is likely solved

closed 2023-01-04 by Hadatko