EmbeddedRPC/erpc #57: Multiple transports

eugene345, 2019-05-16:
Hello,

I have 2 UART interfaces on-board and would like to use eRPC for each. So, I need to have 2 UART transports, 2 clients and (or) 2 servers in a single project. Is it possible to implement? From analysing the eRPC code, seems, not. But maybe there is an workaround?

Best regards,
Eugene.

Hadatko, 2019-05-21:
Hi Eugene. 
It should be possible. But you need create your own erpc_setup_uart_cmsis.cpp. There is one instance of s_transport. Instead of that you need create two of them (s_transport[2] or s_transport_uart1 s_transport_uart2) 

And then you need rewrite a bit init function  erpc_transport_cmsis_uart_init to init them both (or create two init functions for each instance).

eugene345, 2019-05-21:
Hi @Hadatko,

Thanks. This is, seems, easy to create 2 transports.
But how to use 2 transports, if only one client is possible? Assign 1st transport to client and another - to server, no other option? 

Cheers,
Eugene

Hadatko, 2019-05-21:
@eugene345 Yes. Currently you can have several servers, but only one client on one side. We were thinking of adding more clients support with defining which interface belongs to which client (so the interfaces would have to be unique), or rpc call will have one more paramter then now. But this was not priority for us.

ierturk, 2020-07-08:
Hi @Hadatko,

Thanks for the great project `eRPC`. I need more than one `client `within same core as in @eugene345's case. You explained that it is impossible `! Unfortunately you cannot have two clients on one side (at leat in C)` as you said in previous comment. So How can I do this in C++.

Regards,
Ibrahim

Hadatko, 2020-07-08:
Hi @ierturk , I think there is a lot of work behind. Don't have much time to think about that. One solution can be:
At least you need add c++ support here: 
https://github.com/EmbeddedRPC/erpc/tree/develop/erpcgen/src/templates
Where you will generate function implementation inside C++ Client interface (similar as server is doing). When you will create new Client object you will assing necessary objects like transport, msgbufferfactory,...

ierturk, 2020-07-08:
Hi @Hadatko,
Thanks for the prompt reply. I'll check the solution that you recommended.
Regards,
Ibrahim

closed 2023-10-17 by MichalPrincNXP