EmbeddedRPC/erpc #58: Allocate struct on stack

lippit, 2019-06-21:
When using structs with eRPC, they are always allocated with `erpc_malloc`. erpcgen generates the following code for the server:
`data = (someStruct *) erpc_malloc(sizeof(someStruct));`

Why is the struct allocated with malloc, instead of allocating it on the function's stack? When using arrays or primitive types, they are allocated on the stack, as expected:
`uint8_t someArray[10];`

Is there a way to allocate the struct on the stack, is it is done with arrays?

Background of this question is that I want to avoid malloc completely.

MichalPrincNXP, 2019-06-24:
Try to convert struct into array, i.e. use "fooStruct[0] a" instead of "fooStruct a" as the interface function parameter.

amgross, 2022-06-09:
Maybe adding:
`#define erpc_malloc alloca`
Can do the work for whoever interested?

amgross, 2022-06-09:
and need:
`#define erpc_free(x)`
probably