EmbeddedRPC/erpc #59: cannot use namespace and scope resolution operator in .erpc file

jaggi, 2019-06-25:
Hi,
I am trying to include namespaces and to use scope resolution operator in .erpc file. But i m getting error while compiling.Can anyone give a solution to include these two in my IDL file.

Thank you...!

MichalPrincNXP, 2019-06-26:
As for including namespaces, try to do it via including a dedicated header file, use the @include("include_file.h") annotation. As for scope resolution operator in .erpc, I am afraid this is not supported.

jaggi, 2019-06-26:
> As for including namespaces, try to do it via including a dedicated header file, use the @include("include_file.h") annotation. As for scope resolution operator in .erpc, I am afraid this is not supported.

jaggi, 2019-06-26:
Hi MichalPrincNXP,
Thanks for your reply. In my case i have the callback function needed to be implemented in the client side but the callback function and its parameters should be there under my defined namespace.
Can you suggest me how to implement this because i am facing issues while i m using scope resolution operator in .erpc file as callback function to be defined in .erpc file.

Thank you...!

MichalPrincNXP, 2019-06-26:
I am afraid I am not able to help you much, @Hadatko could you please advice?

Hadatko, 2019-06-26:
Hi,
the issue here is that until now the priority was supporting C use cases. I think you want C++ specific feature. C++ specific features are not yet supported.

Hadatko, 2022-06-10:
Hi, i am preparing version where C++ related files will be generated erpc namespace. Is that enough for you or do you need your custom namespace.

Hadatko, 2023-09-27:
i think we understand this issue wrongly. Solution should be implement callback generated by erpc where you can call your callback inside. We will see if we want support more in future, but i think this will be no priority soon.

closed 2023-09-27 by Hadatko