EmbeddedRPC/erpc #60: Where is  erpc_transport_uart_init ?

GabiBarcan, 2019-06-26:
unit_test_server.cpp has the following: 
#if defined(UART)
transport = **erpc_transport_uart_init**(ERPC_BOARD_UART_BASEADDR, ERPC_BOARD_UART_BAUDRATE,
                          CLOCK_GetFreq(ERPC_BOARD_UART_CLKSRC);

In addition to this on the project page:
https://github.com/EmbeddedRPC/erpc
We have the server example using the same call as well:
void example_server(void) {
    // Initialize server running over UART.
    erpc_server_init(
        **erpc_transport_uart_init**(UART0_NonBlocking_Driver);

    // Add the IO service.
    erpc_add_service_to_server(create_IO_service());

    // Run the server.
    erpc_server_run();
}

We tried to use the uart_cmsis but this still needs some work, related to timeout of the read operation and the detection of the new messages, since  **hasMessage(void) { return true; }**
Are we missing any sources ? Was this intentional ?

MichalPrincNXP, 2019-06-27:
Hello @GabiBarcan,
thank you for reporting that. IP-specific UART transports have been replaced by CMSIS UART since 1.4 version. All references to the erpc_transport_uart_xxx or erpc_transport_lpuart_xxx functions need to be removed from the repo (wiki, test code).
If you need some example code showing the CMSIS UART transport usage, visit the https://mcuxpresso.nxp.com page, build the MCUXpressoSDK package for FRDM-K64F board with the multicore optional component, download the package and have a look at boards\frdmk64f\multiprocessor_examples\erpc_[server|client]_matrix_multiply_uart\ example folders.

MichalPrincNXP, 2019-12-20:
Obsolete references to IP-specific UART transport has been removed in [v1.7.3](https://github.com/EmbeddedRPC/erpc/releases/tag/1.7.3)

closed 2019-12-20 by MichalPrincNXP