EmbeddedRPC/erpc #61: Same function name in different interfaces

burgerkingorama, 2019-06-28:
Hello,

when using same function names in different interfaces the C generated files will have duplicate function definitions.

Sample:
```
interface House
{
	open() -> void
	close() -> void
}

interface Car
{
	open() -> void
	close() -> void
}
```

Is there a way of automatically adding a prefix to the generated functions like:
```
void House_open(void);
void Car_open(void);
```

MichalPrincNXP, 2019-06-28:
Hello @burgerkingorama, there is no way to do it and it does not make sense either. When creating the IDL file you are specifying the remotely called function names so the name in IDL and the name in your code should match. If you need to distinguish between open function in different interfaces, put the prefix directly into the IDL (erpc file).

burgerkingorama, 2019-06-28:
Thanks you very much for your quick response!
Of course I was thinking about this workaround to prefix it manually.

But why do you think it doesn't make sense? The functions are somehow grouped by an interface so why do not take the interface name into account? Like C++ classes (or similar languages) do.

To make in optional, I would love to see an annotation like `@interface_function_prefix`. 
So there would be no need to mangle (decorate) the names in the IDL files which leads to much cleaner and readable IDL files.
What do you think?

MichalPrincNXP, 2019-06-28:
It seems to me like a nice-to-have feature, I encourage you to implement that and create the pull request if really needed. Thanks.