EmbeddedRPC/erpc #62: Problems with bidirectional communication

0xff800000, 2019-07-15:
Hello,

I’m trying to setup a bidirectional communication between a M4 core (running FreeRTOS) and a A7 core (running Linux).

When the A7 tries to do a remote call, nothing happens. But when the remote calls are performed from the M4, it gives the error below :

```
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/lib/python2.7/site-packages/erpc/simple_server.py", line 23, in run
    self._receive_request()
  File "/usr/lib/python2.7/site-packages/erpc/simple_server.py", line 31, in _receive_request
    msg = self.transport.receive()
  File "/usr/lib/python2.7/site-packages/erpc/arbitrator.py", line 57, in receive
    info = self._codec.start_read_message()
TypeError: unbound method start_read_message() must be called with BasicCodec instance as first argument (got nothing instead)
```
Could you please help me ?

Here is the M4 code in C:
```C
void functionsM4_eRPC(...){
	...
} 

void poll_server(void *param){
    PRINTF("Polling task started\n\r");
    while(1) erpc_server_poll();
}

void app_task(void *param){
	erpc_mbf_t message_buffer_factory;
	transport = erpc_transport_rpmsg_lite_tty_rtos_remote_init(101, 1024, (void*) RPMSG_LITE_SHMEM_BASE, ERPC_TRANSPORT_RPMSG_LITE_LINK_ID, NULL, RPMSG_LITE_NS_ANNOUNCE_STRING);
	message_buffer_factory = erpc_mbf_rpmsg_tty_init(transport);
	erpc_transport_t arbitrator = erpc_arbitrated_client_init(transport, message_buffer_factory);
	erpc_server_init(arbitrator, message_buffer_factory);
	erpc_add_service_to_server(create_linux2rtos_service());

	if(xTaskCreate(poll_server, "APP_POLL", APP_TASK_STACK_SIZE, NULL, tskIDLE_PRIORITY + 1, &app_task_handle) != pdPASS) {
		PRINTF("\r\nFailed to create polling task.\r\n");
		return;
	}

	while(1){
		functionsA7_eRPC(...);
	}
}
```
Here is the A7 code in Python :
```Python
class rtos2linuxServiceHandler(erpc_message.interface.Irtos2linux):
    def functionsA7_eRPC(self, ...):
        ...

def runServer(transport):
    server = erpc.simple_server.ServerThread(transport, erpc.basic_codec.BasicCodec)
    msgHandler = rtos2linuxServiceHandler()
    msgService = erpc_message.server.rtos2linuxService(msgHandler)
    server.add_service(msgService)
    server.start()
    print('\r\neRPC server started\r\n')

def runClient(transport):
    clientManager = erpc.client.ClientManager(transport, erpc.basic_codec.BasicCodec)
    client = erpc_message_M4server.client.linux2rtosClient(clientManager)
    while True:
        client.funcitonsM4_eRPC():

if __name__ == "__main__":
    transport = erpc.transport.SerialTransport("/dev/ttyRPMSG101", 115200)
    arbitrator = erpc.arbitrator.TransportArbitrator(transport, erpc.basic_codec.BasicCodec)
    print('Selected ttyRPMSG Transport')
    runServer(arbitrator)
    runClient(arbitrator)
```

MichalPrincNXP, 2019-07-16:
Hello @0xff800000,
I have never tried to run bidirectional communication between CM7/Linux/Phyton and CM4/FreeRTOS/C, but I think you should have the server and the client task separated on the CM4 side (app_task in your code combines both together). Attached is the C source code for both sides of the bidirectional communication that works on our platforms and could be used as reference for you. You should be able to setup your erpc application accordingly. Hope it will help you.
Regards
Michal

[erpc_two_way_rpc_rpmsg_rtos.zip](https://github.com/EmbeddedRPC/erpc/files/3397021/erpc_two_way_rpc_rpmsg_rtos.zip)

0xff800000, 2019-07-17:
Hello @MichalPrincNXP ,
I tried to modify the M4 code as you suggested however it gives the same error. My guess is that the problem comes from my Python code.
Do you have an example code (either Linux/Python or Linux/C) that uses the Transport Arbitrator ?
Thanks,
0xff800000

MichalPrincNXP, 2019-07-17:
Hello @0xff800000,
I recommend to visit the https://mcuxpresso.nxp.com/ and build and download the sdk package for FRDM-K64F board (select all optional components once being on the web page to include the multicore support). There is the multiprocessor erpc_remote_control examples placed in <your package.zip>boards\frdmk64f\multiprocessor_examples folder then. This example implements the transport arbitrator and can serve as another reference for you. The python app. with arbitrator is attached.

[run.zip](https://github.com/EmbeddedRPC/erpc/files/3402463/run.zip)

Regards
Michal

0xff800000, 2019-07-18:
Hello @MichalPrincNXP ,
I read the python example you sent and I could find the errors in my code.
Thank you very much for your help !
0xff800000