EmbeddedRPC/erpc #63: problem in communication using windows as a client and linux as server

bharathkotari, 2019-07-15:
Hi,

We are using erpc server on linux and erpc Client on Windows.
(Erpc client on windows is compiled using migw)

We observed that the custom functions called from the client are not executed from server.
We have added the function details on .erpc also
If there are no custom functions and try to run the test_arbitrator sample code , it works fine. 
The addition of the custom code is making the system to stop at the call of function. (There is no segmentation fault)
Reported).
Attached is the .erpc and custom function in client. 
this is test_arbitrator code

**test_arbitrator.erpc**
```
/*!
 * Copyright (c) 2016, Freescale Semiconductor, Inc.
 * Copyright 2016 NXP
 * All rights reserved.
 *
 * SPDX-License-Identifier: BSD-3-Clause
 */

@crc
@output_dir("erpc_outputs/")
program test;

import "../common/unit_test_common.erpc"

@group("firstInterface")
interface FirstInterface
{
    whenReady() -> void
    oneway firstSendInt(int32 a)
    firstReceiveInt() -> int32 
    ####################################
    ourCustomFun(some datatype) -> returnvalue
    ####################################
    stopSecondSide() -> void
    getResultFromSecondSide() -> int32
    oneway quitFirstInterfaceServer()
    nestedCallTest() -> int32
    @nested
    callSecondSide() -> int32
}

@group("secondInterface")
interface SecondInterface
{
    oneway secondSendInt(int32 a)
    secondReceiveInt() -> int32
    oneway quitSecondInterfaceServer()
    oneway enableFirstSide()
    @nested
    callFirstSide() -> int32
}

```

MichalPrincNXP, 2019-07-17:
Hello @bharathkotari,
I have tried to add some custom function into the test_arbitrator.erpc, regenerated the shim code, added the custom function implementation into the side with other FirstInterface function implementations and called this function from the oposite side. All works fine using the multicore environment and the rpmsg-lite transport layer.
What transport layer are you using? Have you added the newly introduced custom function implementation into the right side of the erpc system? When using the erpc arbitrator we have both the erpc server and client on both sides of the system. If you would like to try just the single client - single server setup I would recommend any other test (not using the arbitrator).
Regards
Michal

bharathkotari, 2019-07-17:
Hi@MichalPrincNXP ,
thank you for the reply, we have not explictly specified the transport layer, and we are using erpc_tcp_transport and  it is the default one. we added custom function to the FirstInterface and implemented it in the server, In both the machines. when we are calling our custom function, the windows client is able to establish a connection with the linux server but stopping at the functioncall. if we revert back our changes to default test arbitrator, the windows client is able to connect to linux server and performed tests.
here we are facing some problems

- Even if we make slight change in .erpc file communication is not happening(like changing the return type, or changing the parameters)

- But the same is working fine in linux-to-linux (custom functions with different return values)

- The default test_arbitrator is working with windows-to-linux, if we make any change in .erpc it is not working

regards,
Bharath kotari

MichalPrincNXP, 2019-07-17:
Hi @bharathkotari,
this seems to me like you are not regenerating the shim code after the erpc file change. Could you confirm that both the client and the server shim codes are regenerated and rebuilt correctly before running the application?

Regards
Michal

bharathkotari, 2019-07-18:
Hi @MichalPrincNXP , after some through comparisation we found out that crc is not same. we kept the same crc in both the machines, and it worked.

Thankyou for your support

Regards 
Bharath kotari

Hadatko, 2019-07-21:
Hi @bharathkotari , just notice: When crc is not the same you are likely not using same version (if you are using generated crc). Even empty space in IDL and regenerating shimcode will change CRC value.

closed 2019-10-31 by Hadatko