EmbeddedRPC/erpc #64: multiple client request not able to process

jaggi, 2019-07-17:
Hi,
I am trying to process multiple request of client but server is not able to do it.Can anyone suggest me to process multiple requests of client.

Thanks

MichalPrincNXP, 2019-07-17:
Hi @jaggi,
I guess you have multiple tasks on the client side and would like to request erpc calls from these tasks, have I captured it correctly? If so, the erpc calls/requests needs to be serialized somehow. You could disable tasks switching when the erpc request is issued, for instance (not so nice as the eprc call could take some time). Or, you have to change the application logic to implement just one erpc client task and ensure data exchange between the client task and other task by other RTOS-provided means like messages, queues, etc. Hope it helps.
Regards
Michal

jaggi, 2019-07-17:
Hi MichalPrincNXP ,
Thank you for the reply.
In my scenario, i will be running both server and client. When the server receives a request from the client it process the request and both server and client terminates. What i need is server should not terminate after it process the request instead it should wait for another client request . So when i run client once again it should process even that without runnuing server once again.  So help me in doing this.

Thanks

MichalPrincNXP, 2019-07-17:
Hi @jaggi,
what is the reason of both server and client termination? Normally, the server should not terminate unless you explicitly request that using a dedicated erpc function (quit) that shutdowns the erpc server, or there is some error on the server side and the erpc server is closed by itself. I guess the second case happens on your side.
It seems your erpc system is not properly setup and I would recommend to debug what happens when the erpc request reaches the server side.
If you would like to see an erpc example, I recommend to visit the https://mcuxpresso.nxp.com/ and build and download the sdk package for LPCXpresso54114 boards. Multicore erpc examples are placed in board/lpcxpresso54114/multicore_examples folder then.

Regards
Michal

Hadatko, 2019-07-21:
Hi @jaggi , are you using C or Python? Also do you use erpc_server_poll or erpc_server_run function in case if you are using C.

Hadatko, 2022-06-10:
No responds so far. Closing

closed 2022-06-10 by Hadatko