EmbeddedRPC/erpc #65: Memory issue when using @retain

burgerkingorama, 2019-08-02:
We're using `@retain` on a list output parameter.

IDL:
```
struct result { ... }

interface test {
    transferResults(out list<result> @retain) -> void
}
```

In the server implementation we do something like:
```
void transferResults(list_result_t_1_t * results) {
    results->elementsCount = g_resultsCount;
    results->elements = &g_results; // static element, do not free
}
```

Now we realized, that this will leak memory, because when using `@retain` not only the list elements are not freed but the struct itself (`*results`) will also not be freed.

In the shim code is a `erpc_alloc` call and we as a server implementer have no change to free it after it was send over erpc.

For the list elements is obviously okay to not be free, because they are managed by the implementer itself. That was our understanding of  `@retain`.

Are we using it with a wrong expectation or is they a way for us to call `erpc_free` on the struct later or do you agree it is a issue in the generated shim-code?

Hadatko, 2019-08-09:
Hi @burgerkingorama,
@retain was created for special usecases when somebody don't want free allocated structures at all. E.g. he can pass allocated structure to global variabel for using it during other calls (or passing to other tasks). Then he need to free whole memory.

burgerkingorama, 2019-08-10:
Hi @Hadatko!

We are using  exactly for that special usecase. Please see my sample above.

Still there is memory that is owned and can only be managed by eRPC, which is not free by eRPC and a user is not able to free. This will leak memory because on every call of that function eRPC allocates a new struct that nobody (except for eRPC) can free again.

If you like, I can show you my concerns in the shim-code.

Thanks,
bko

Hadatko, 2019-08-10:
@burgerkingorama. C doesn't have list. For that reason list here means struct. And for that reason @retain is applied to struct. And for that reason you should free memory same way as eRPC allocated it. 

But you can try something like this:
interface test {
    transferResults(out list<result> @retain @length(len) @max_length(10), uint32 len ) -> void
}
But i see that test code is commented, so i am not sure if this will work.

burgerkingorama, 2019-08-11:
Hi @Hadatko,

>  And for that reason you should free memory same way as eRPC allocated it.

That's what I'm saying. It is not possible to free the struct that was allocated by eRPC, because it is still needed after the implementation of the RPC function is left (means proceeding shim code).

The thing is, we are talking about a `out` parameter. For a `in` parameter you are of course right. We can free it directly inside of the RPC implementation. For `out` we can not, because the parameter is needed by eRPC.

Of course you can store the pointer and try to free later but that's seems to be very unpractical.

Hadatko, 2019-08-11:
@burgerkingorama i agree it is unpractical, but currently you need count with that. @retain was created for that purpose you want free these data later. 

So as i understand what you want to achieve now. 
You have two options: 
1. do not use @retain and you need allocate your list same way as eRPC would do. This way eRPC will free the memory for you.
2. Currently i am not sure what is best case for you and if it will work but theoreticaly you can do this
2.a: transferResults(out list @retain @Length(len) @max_length(10), uint32 len ) -> void
2.b:
struct myStruct{
}