EmbeddedRPC/erpc #66: Codec issue when transfereing lists with size 0

burgerkingorama, 2019-08-06:
When transferring an empty list, the codec will be confused and all following structs are corrupted.

IDL:
```
interface Test {
	getData(out list<uint8>) -> void
}
```

The client shim code looks like this:
```
static void read_list_uint8_1_t_struct(erpc::Codec * codec, list_uint8_1_t * data) {
	codec->startReadList(&data->elementsCount);
	data->elements = (uint8_t *) erpc_malloc(data->elementsCount * sizeof(uint8_t));
	if (data->elements == NULL)
	{
		codec->updateStatus(kErpcStatus_MemoryError);
	}
[...]
```

Depending on the implementation of `erpc_malloc` the function may return `NULL` if the size to malloc is zero.  

In the `NULL` case, which is our case is not an error the codec updates its state to error and all following struct are corrupted then.

MichalPrincNXP, 2020-04-21:
Hello @burgerkingorama , thanks for reporting this, providing the fix and sorry for not being responded yet. This seems to me like a valid use case but need the confirmation from @Hadatko . Is it allowed to use lists as out parameters? If so, we should handle the use case with empty list being parsed as proposed in #72 . @Hadatko , could you briefly comment, please? Thank you.

Hadatko, 2020-05-08:
I think the condition if (data->elements == NULL) should be changed to 
 if (data->elements == NULL && data->elementsCount>0)

Hadatko, 2020-05-08:
Also need be confirmed that if if elementsCount is set to 0 then write list struct will send 0 to other side and nothing more. I would suggest to add tests for binary/list type for this cases.

closed 2022-02-28 by MichalPrincNXP