EmbeddedRPC/erpc #69: Memory violation in free_*_struct function when using unions

burgerkingorama, 2019-09-09:
Hello!

When using an `union` within a struct the free-function does not handle all switch-cases but only those that needs special sub-free-calls like `list<>` etc. Unfortunately this free calls is then always performed regardless of which union type was used at that moment.

Example eRPC IDL:
```
struct TestStruct
{
	int32 handle_count // uint32 handle_count causes error in eRPC code
    union(handle_count)
	{
	case 0:
		uint16 start_handle
		uint16 end_handle
	case 1:
		uint16 handle
		uint16 offset
	default:
		list<uint16> handles
	} range
}
```

The free call looks like this:
```
static void free_TestStruct_struct(TestStruct * data)
{
    switch (data->handle_count)
    {
        default:
        {
            free_list_uint16_1_t_struct(&data->range.handles);
            break;
        }
    }
}
```

This will lead to memory violation if the list was not in use.

Hadatko, 2019-09-09:
Hi @burgerkingorama, 
thank you for your interest. For code size /readability optimization would be better to revert your commit changes and add new for loop for cases which don't need to free data. 
I think your output will be:

static void free_TestStruct_struct(TestStruct * data)
{
    switch (data->handle_count)
    {
	case 0:
		break;
	case 1:
		break;
       ....
       case n:
             break;
        default:
        {
            free_list_uint16_1_t_struct(&data->range.handles);
            break;
        }
    }
}

Suggested:
static void free_TestStruct_struct(TestStruct * data)
{
    switch (data->handle_count)
    {
	case 0:
	case 1:
        ....
        case n:
		break;
        default:
        {
            free_list_uint16_1_t_struct(&data->range.handles);
            break;
        }
    }
}

MichalPrincNXP, 2022-01-20:
resolved in #228

closed 2022-01-20 by MichalPrincNXP