EmbeddedRPC/erpc #74: Example on Linux

bhaktatejas922, 2019-10-05:
Hi, 
I am pretty new to RPC/RPMsg and was wondering if you had any guidance on how to get a simple example setup on Linux? 

I just wanted to send and receive a message through stdout/stdin and have been trying to get  the example here to work: https://github.com/EmbeddedRPC/erpc/wiki/Getting-Started

Do you have any advice on how to handle the transport? Or are there any other examples/documentation somewhere. Apologies if this is a very simple question, I'm a bit of a beginner. Thanks!

Hadatko, 2019-10-06:
Hi the link you provided has everything to create client/server application. For adding new transport you can read here: https://github.com/EmbeddedRPC/erpc/wiki/Porting-Guide. This second page is maybe not very descriptive but you can found examples here: https://github.com/EmbeddedRPC/erpc/tree/master/erpc_c/transports
and setup functions here: https://github.com/EmbeddedRPC/erpc/tree/master/erpc_c/setup

It would be good to write which programming language are you using (python/C?). 

Basically you need implement send/receive function (should be straightforward). Both need to be (at least in first implementation) blocking. So send will block application until message is send and receive until message is received. So you will need likely something like system mutex. 

Other thing i can suggest are tests. Tests for eRPC are written for usage on Linux. Transport layer is tcp. You need only compile code and then run server app and then client app.

bhaktatejas922, 2019-10-06:
Sorry I forgot to mention the language. I am using C. I'll look into the tests. Your response was very helpful, thank you!

bhaktatejas922, 2019-10-07:
Hi Hadatko,

Thanks for the advice, got the tests to work. Is there a way you know of to send a statically allocated struct? I saw your response here: https://github.com/EmbeddedRPC/erpc/issues/23 does this mean there is no way to send something statically allocated as of now?

Hadatko, 2019-10-07:
Hi @bhaktatejas922 , please take a look here: #58. Is this suttisfying your question?

bhaktatejas922, 2019-10-08:
Not exactly. I need to send a statically allocated struct (not an array), and I tried to do so with some modifications to the test_struct test. I tried to use TestString1 and statically allocate the primate struct on the client side as so in test_struct_client_impl.cpp `TEST(test_struct, TestString1)
{
    static primate prim8;
    char str[] = "Ape";
    prim8.species = &str[0];
    prim8.is_ape = false;
    EXPECT_TRUE(checkString(&prim8) == 0);
}`

but the tests don't pass for this and it leaks on the server side and expects 2 values to be freed. I suspect there must be some internal erpc_malloc call I am missing that causes this. 

Do you know of a better way to go about sending a statically allocated struct, or is it not possible at this time? I strictly need to send a statically allocated struct. Thanks!

Hadatko, 2019-10-08:
I don't see reason why it would not work with only this modification. I did same and tests passed.

Hadatko, 2019-10-08:
[image]

bhaktatejas922, 2019-10-08:
The checkString function in test_struct_server_impl.cpp calls erpc_free twice, and if you comment them out, you get the memory leak error

 `int32_t checkString(const primate *p)`
`{`
    `int b = strcmp(p->species, "Ape");`
    `printf("%s", p->species);`
   ` //erpc_free(p->species);`
  `  //erpc_free((void *)p);`
 `   return b;`
`}`
[image]

Hadatko, 2019-10-08:
It doesn't matter how you are allocated space on client side. Server is always doing same allocation. It is based on IDL. Currently only dynamic i think.

bhaktatejas922, 2019-10-15:
Any plans to have an implementation for statically allocated on server side? Embedded systems would have a good use for this.

Hadatko, 2019-10-15:
@MichalPrincNXP I will let answer on you.

MichalPrincNXP, 2019-10-15:
Hi @bhaktatejas922 ,
you are correct that all struct allocations are dynamic in the generated server shim code. There is no option to change that unless you modify the shim code manually, which is not desired.
I will put this requirement into the list of improvements but do not promise to implement it soon. If you would be willing to contribute to this project and prepare the pull request we would appreciate it. Thank you.