EmbeddedRPC/erpc #75: Porting eRPC to arm linux platform

maningst, 2019-10-11:
Hello

When I read the github code, I found the following variables used:
//! @brief Opaque transport object type.
typedef struct ErpcTransport *erpc_transport_t;
Where do you define this? if is a hidden structure, what is the object file or linker library to contain it?(is it compatible with arm-gcc compiler?)

So far my understanding of the general porting procedure is:
1. On Linux host PC, create the necessary IDL file and generate the shim code by using Linux erpcgen tool(which can be compiled from linux NATIVE gcc/g++ compiler).
2. Copy all files inside erpc_c folder into arm target folder, use the arm-gcc/arm-g++ compiler  to build the libespc.a file and install necessary header files
3. Together with the shim code, write own client and server application files, which can call proper init and service routine.

Can you please correct me if I was wrong? By the way do you have any plan to have erpc_setup_xxx implementation for tcp? Thank you

Regards

Hadatko, 2019-10-11:
Hi @maningst 
This ErpcTransport is not defined. Compiler is expecting pointer type and compilation will pass. It something like `void *` with name. 
1. Generating shim code can be done on Windows/Linux/Mac. It doesn't matter. 
2. It doesn't matter where the files will be coppied. You need add proper include paths for your compiler. You can generate liberpc library or you can use sources directly during compilation.
3. right

P.s.: I think we have pullrequest which has erpc_setup_... for tcp. You can reused that one if it will match your needs: #35

maningst, 2019-10-11:
OK, thanks a lot for your quick reply.

Regarding the common files inside "erpc_c" folder, I can select a few of them(necessary one) to compile, right? 
May I know what is the difference between "erpc_arbitrated_client_init" and "erpc_client_init"? which one should be used for TCP? From I see although both "erpc_client_setup.cpp" and "erpc_arbitrated_client_setup.cpp" are located under "infra" folder, I can only compile one of them at a time into the liberpc.a, because these two files define and use same global client variables.

Thanks

Hadatko, 2019-10-11:
Hi,
1. Yes
2. Difference is that erpc_client_init is simpler more transparent. It is used for communication from client to server side (and back if any return values). erpc_arbitrated_client_init is used when you want asynchronous clients calls from both sides (clients and servers are on both sides of communication)

maningst, 2019-10-14:
Hello

In order to deal with the requirement of bidirectional remote_function() call,  such as one side triggers calling remote_function 1 & 2, and the other side triggers calling remote_function 3 & 4,  is it possible to use this client and server routine on one Transport template? If yes, what is the normal procedure to do the initialization on both client/server sides? Sorry I did not find any examples on the wiki page about it.

Best Regards

Hadatko, 2019-10-14:
Hi,
please take a look to this test: https://github.com/EmbeddedRPC/erpc/tree/master/test/test_arbitrator

maningst, 2019-10-15:
Hello

When I execute following code on TCP client/server communication, the remote_func calling works fine, but during server terminate period "s_server.destroy()" inside erpc_server_deinit() always complains "free(): invalid pointer", any clue or experience on this?
   
erpc_transport_t transport = erpc_transport_tcp_init("localhost", 12345, true);
if(transport == NULL) {
    printf("erpc_transport_tcp_init failed \n");
}
/* MessageBufferFactory initialization */
erpc_mbf_t message_buffer_factory = erpc_mbf_dynamic_init();

/* eRPC server side initialization */
erpc_server_init(transport, message_buffer_factory);
erpc_add_service_to_server(create_remoteFunc_service());

/* run server */
erpc_server_run();
	
//erpc_transport_tcp_deinit();
erpc_server_deinit();

Regards

Hadatko, 2019-10-15:
May i see your IDL file?

maningst, 2019-10-15:
Sure, so far I just use it for some basic testing:

/*!
 * You can write copyrights rules here. These rules will be copied into the outputs.
 */
@outputDir("erpc_outputs") // output directory
program erpc_remote_func_test; // specify name of output files

const int32 sample_macro = 5;

interface remoteFunc // cover functions for same topic
{
    remote_Func1_test_add(in int32 para1, in int32 para2, out int32 out_para) -> int32
    remote_Func2_test_minus(in int32 para1, in int32 para2, out int32 out_para) -> int32
    remote_Func3_test_multiply(in int32 para1, in int32 para2, out int32 out_para) -> int32
    remote_Func4_test_divide(in int32 para1, in int32 para2, out int32 out_para) -> int32
}

maningst, 2019-10-15:
[app.zip](https://github.com/EmbeddedRPC/erpc/files/3727526/app.zip)
Testing application as attached

Hadatko, 2019-10-22:
File is not accessible for me.

maningst, 2020-01-07:
Hello

Sorry for the late reply, did you manage to find something related to "free(): invalid pointer" issue inside  erpc_server_deinit()?

Regards

Hadatko, 2020-01-07:
Hi @maningst could you try new version of erpc 1.7.3 released recently? If the issue is in erpc_server_deinit it is likely already fixed. There were staticly allocated services which were freed during deinitialization.

maningst, 2020-01-09:
Yes, it is already fixed in the new release v1.7.3.  thank you for the support

closed 2020-01-09 by maningst

Hadatko, 2020-01-09:
You are welcome. I am glad that we have fix already upstreamed for you.