EmbeddedRPC/erpc #79: porting eRPC for baremetal

tmordeko, 2019-10-28:
Hi,

Started looking at eRPC for remote procedure implementation between two cores over uart channel.

One core has no OS running, is it possible to run eRPC client and/or server on a baremetal core without OS running ? Is it in analogy to "no threads" or threading disabled define (in erpc_config_internal.h) ? as class TransportArbitrator has both mutex and semaphore structures.

/Tzvika

Hadatko, 2019-10-28:
Hi @tmordeko , currently i think for BM you can have only client on one side and server on another side. You can't use arbitrator feature (have combination of multiple clients/servers on one side).

tmordeko, 2019-10-28:
Thanks for your prompt reply !

I am running into following compilation error:
erpc/erpc_c/infra\erpc_transport_arbitrator.h:110:9: error: unknown type name 'Semaphore'
        Semaphore m_sem;           /*!< Client semaphore. */
        ^
how do you suggest to resolve it ?

Can you please elaborate on 'arbitrator feature', looked at the wiki and still not sure I understand it, is it possible to disable it ?

/Tzvika

Hadatko, 2019-10-28:
Do not include arbitrator code into your app. Are you using library? Do not use includes and functions call for arbitrator. Also you should set in your erpc_config.h BM Macro instead of any OS macro.

Hadatko, 2019-10-28:
I think default should be fine. Please read this articel: https://github.com/EmbeddedRPC/erpc/wiki/Getting-Started

tmordeko, 2019-10-28:
I read the wiki, it doesn't talk how to disable arbitrator code :-)
there are indeed some files in erpc directory that contain arbitrator. as you mentioned should I exclude them from the build ?

Hadatko, 2019-10-28:
Link i provided describes files you need for basic app. On wiki is also section PORTING which describes what are other files used for. 
Please exclude files from build which are not necassary for your application. 
@MichalPrincNXP maybe we can consider add to arbitrator related files macro for excluding code from compilation if OS not presented. I am not sure if i like it, we need to see impacts.

MichalPrincNXP, 2020-04-17:
@Hadatko : The following check will be added into arbitrator code to prevent from build in no-threading configuration in coming release.
```
#if ERPC_THREADS_IS(NONE)
    #error "Arbitrator code does not work in no-threading configuration."
#endif

```
@tmordeko : Can we close this issue? Or, is there still something unclear, to be improved in doc?

tmordeko, 2020-04-19:
@Hadatko yes thanks

closed 2020-04-19 by tmordeko