EmbeddedRPC/erpc #80: Client/Server implementation from the same core

tmordeko, 2019-11-04:
Hi,

We are evaluating eRPC for our project. Is it possible to support both eRPC client and eRPC server on same core simultaneously (client/server invoking/serving different calls) over same UART connection (since we have only single UART line between the cores) ?

Thanks in advance,
Tzvika

MichalPrincNXP, 2019-11-06:
Hello @tmordeko, theoretically, this use case should work but we have not tried to be honest. There is the TransportArbitrator that allows to share transport between client and server.
Regards
Michal

tmordeko, 2019-11-24:
Thanks Michal for your response.

Can you please explain how should I calibrate my code / erpc library to work with TransportArbitrator (sorry in advance I am not c++ expert). An example if you have one could help a lot.

Is there another method, even more naive way to make it work, such that each side could be both client and server over the same uart ?

Thanks,
Tzvika

MichalPrincNXP, 2019-11-25:
Hello @tmordeko ,
erpc examples are included in NXP MCUXpressoSDK packages. These can be obtained through the https://mcuxpresso.nxp.com/ webpage. You can select for instance LPCXpresso54114 sdk. After downloading the package, you can have a look at erpc_two_way_rpc_rpmsg_rtos example located in boards/<board name>/multicore_examples/erpc* folder. This example shows the eRPC transport arbitrator usage. The rpmsg_lite transport is used in this case, you have to consider replacing the transport layer by the UART then. Attached are app. sources for both sides, hope it helps.
Regards
Michal

[erpc_arbitr.zip](https://github.com/EmbeddedRPC/erpc/files/3886083/erpc_arbitr.zip)

closed 2022-06-10 by Hadatko