EmbeddedRPC/erpc #83: transport is not working on Linux if debug logging is disabled

SASIDHARAN-dot, 2019-12-13:
After deep level of debugging, I figured out the accept() call is failed when client try to connect to the server. Then I changed the accept(serverSocket, &incomingAddress, &incomingAddressLength) call parameter to accept(serverSocket, 0, 0). Then it worked. Can you initialize the incomingAddress and incomingAddressLength to 0, and then pass as reference to accept() function. is it possible to initialize all the local function variable to zero as will as autogen variable?

MichalPrincNXP, 2020-01-21:
Hello @SASIDHARAN-dot 
could you please provide your setup when the TCP transport is not working? I have tried to run erpc test in Linux (ubuntu 16.04) and does not face any issue. Thank you.

MichalPrincNXP, 2020-02-28:
Hello @SASIDHARAN-dot  , did you have a chance to read my last comment? Thank you.

SASIDHARAN-dot, 2020-03-16:
yes. It didn't fail in linux may be its a high end processor. I tried on stm32 microcontroller it didn't work.  Ensure  setting this flag to zero #define TCP_TRANSPORT_DEBUG_LOG (0).

MichalPrincNXP, 2020-04-09:
Hello @SASIDHARAN-dot ,
I am not able to reproduce the issue. Even when #define TCP_TRANSPORT_DEBUG_LOG (0) the erpc tests in Linux work fine.
Could you kindly describe your use case? Have you setup the erpc communication between a PC running Linux and an MCU with Ethernet capability? What TCP/IP stack have been used on the MCU side? 
Or, have you been running erpc between two MPUs with embedded Linux running on them? 
Thank you for any details.

Hadatko, 2020-04-10:
FYI: Similar/Same issue had also Chris Reed. If debug log were disabled, tcp stopped working for him.  But it was many times ago and later i couldn't reproduce issue so i thought that it was fixed.

SASIDHARAN-dot, 2020-05-29:
Hello Michal,
The use case is communication between PC and MCU through USART. and the POSIX version for tcp stack is  POSIX.1-2001

Arunram241196, 2020-06-05:
Hi @SASIDHARAN-dot how many cpu cycles does erpc consumes??