EmbeddedRPC/erpc #86: Unable to declare callback functions one by one next to another, which of the type with return value.

zuzy, 2019-12-18:
``` c#
/*!
 * Copyright 2017 NXP
 * All rights reserved.
 *
 * SPDX-License-Identifier: BSD-3-Clause
 */

@output_dir("erpc_outputs")
program test
import "../common/unit_test_common.erpc"

oneway callback1_t(int32 a, int32 b)
oneway callback2_t(int32, int32)
callback3_t(int32 a, int32 b) -> int32

@include("test_core1.h")
@group("core0")
interface ClientCore0Services
{
    myFun(in callback1_t pCallback1_t, out callback1_t pCallback2_t) -> void
    myFun2(callback2_t pCallback1_t, out callback2_t pCallback2_t) -> void
    testFun(in callback3_t p_callback, int32 arr1, int32 arr2) -> int32;

    @include("callbacks1.h")
    callback1_t callback1;

    // callback3_t callback4
    callback1_t cb_add1
    callback1_t cb_add2
    callback3_t cb_add
    callback3_t cb_mul
    callback3_t cb_sub
    callback3_t cb_div

    @include("callbacks1.h")
    callback2_t callback2(param1, param2);
}
```

I changed code in `test_callbacks folder` like this. But erpcgen the `client.cpp` with a extra return at the tail of implement functions, which stops me compiling. I just delete the returns manually, then the test works.

And it works if I declare only one callback in the interface.

``` c
void cb_add2(int32_t a, int32_t b)
{
    callback1_t_core0(kClientCore0Services_service_id, kClientCore0Services_cb_add2_id, a, b);
}

// ClientCore0Services interface cb_add function client shim.
int32_t cb_add(int32_t a, int32_t b)
{
    return callback3_t_core0(kClientCore0Services_service_id, kClientCore0Services_cb_add_id, a, b);
return;
}

// ClientCore0Services interface cb_mul function client shim.
int32_t cb_mul(int32_t a, int32_t b)
{
    return callback3_t_core0(kClientCore0Services_service_id, kClientCore0Services_cb_mul_id, a, b);
return;
}

```

Hadatko, 2019-12-18:
I am confirming issue with latest erpcgen: Generated by erpcgen 1.7.2 on Wed Dec 18 08:15:37 2019.

Hadatko, 2022-02-05:
Not an issue anymore.

closed 2022-02-05 by Hadatko