EmbeddedRPC/erpc #89: eRPC C++ Client-Server Matrix Multiply - client and server not communicating

hale2bopp, 2020-03-05:
**Specs**
eRPC Version : 
System : MacOS Mojave 
Transport used : TCP (with erpc_setup_tcp.cpp used from someone else who updated the same with a pull request)
Message buffer: static

**Problem**
I have gotten the system to build and am trying to run the client and server on my computer, which means it is ported to unix, and using the TCP transport. I have tried the matrix multiply from the python example and used the same IP address and port and that works, and I can see the packets on wireshark for that port. When I try the same thing for the C++ example, I am not able to do so.

I first built the erpc project so that all the object files were available in Debug/Darwin. I used these to compile the final object files for the server and client. (This is more of a proof of concept which is why I haven't yet written a makefile or a CMake file to compile the project, I want to be sure that it will work before making that effort).

I have attached the IDL files, main.c that I have used. I used erpcgen to generate the shim code for the same, and I am also adding the manual list of compile commands to follow in order to compile the server_main.cpp, client_main.cpp and the final server and client object files. 

I am using the same IDL file as the python example, and erpcgen generates the .cpp files required for me with the code to do the matrix multiplication etc. I wrote a main file for the server and client which specifies the transport, message buffer, initializes the server/client, and opens them up for connection. 

server_main:
`int main(void)
{
    erpc_transport_t transport = erpc_transport_tcp_init("127.0.0.0", 50000);	

//    erpc_mbf_t message_buffer_factory = erpc_mbf_rpmsg_init(transport);
    erpc_mbf_t message_buffer_factory = erpc_mbf_static_init();

    /* eRPC server side initialization */
    erpc_server_init(transport, message_buffer_factory);

    printf("eRPC intialized\r\n");

    /* adding the service to the server */
    erpc_add_service_to_server(create_MatrixMultiplyService_service());

    printf("MatrixMultiply service added\r\n");

    /* run server */
    while(1)
	    erpc_server_run();

//    while (1)
//    {
//        /* process message */
//        erpc_server_poll();
//    }
    return 0;
}
`

client_main:

`
int main()
{
    /* Matrices definitions */
    Matrix matrix1, matrix2, result_matrix = {{1}};

    /* init eRPC client environment */
    /* TCP transport layer initialization */
    erpc_transport_t transport = erpc_transport_tcp_init("127.0.0.0", 50000);	

    /* MessageBufferFactory initialization */
    erpc_mbf_t message_buffer_factory = erpc_mbf_dynamic_init();
    
    /* other code like print result matrix */
    printf("Initializing client side\n");

    /* eRPC client side initialization */
    erpc_client_init(transport, message_buffer_factory);

    /* other code like init matrix1 and matrix2 values */
    
    int i,j;
   
    for (i=0;i<matrix_size;i++)
	for (j=0;j<matrix_size;j++)
		matrix1[i][j] = 1;

    for (i=0;i<matrix_size;i++)
	for(j=0;j<matrix_size;j++)
		matrix2[i][j] = 2;

    /* other code like print result matrix */
    printf("Calling eRPC matrix multiply on server side\n");

    while(1){	
    /* call eRPC functions */
    erpcMatrixMultiply(matrix1, matrix2, result_matrix);

    /* other code like print result matrix */
    printf("Here is the result of the matrix multiplication: \n");

    for (i=0;i<matrix_size;i++){
        for(j=0;j<matrix_size;j++)
                printf("result_matrix[%d][%d] = %d\t",i,j,result_matrix[i][j]);
	printf("\n");    
    }
    }
    /* eRPC client side initialization */
    erpc_client_deinit();

    return 0;
}`

IDL file : 

`program erpc_matrix_multiply

/*! This const defines the matrix size. The value has to be the same as the 
    Matrix array dimension. Do not forget to re-generate the erpc code once the 
    matrix size is changed in the erpc file */
const int32 matrix_size = 5;

/*! This is the matrix array type. The dimension has to be the same as the 
    matrix size const. Do not forget to re-generate the erpc code once the 
    matrix size is changed in the erpc file */
type Matrix = int32[matrix_size][matrix_size];

interface MatrixMultiplyService {
    erpcMatrixMultiply(in Matrix matrix1, in Matrix matrix2, out Matrix result_matrix) -> void
}
`

Instructions to build: 

`1. make in erpc folder
2. make a folder with the erpc_matrix_multiply.erpc
3. erpcgen on this to generate the shim code in service/erpc_matrix_multiply

After that, 

Server-Side commands

1. Step one: make in erpc folder to generate obj files in Debug/Darwin/erpc/obj/erpc_c

2. Compile server.o which is made from erpc_matrix_multiply_server.cpp

# Cmd to compile server.o 
g++ -I../erpc_c/infra -I../erpc_c/port -I../erpc_c/transports -I../erpc_c/setup  -I../erpc_c/config -c service/erpc_matrix_multiply/erpc_matrix_multiply_server.cpp -o server.o -ggdb

3. Compiler server_main.o

# Cmd to compile server main 
gcc -I../erpc_c/infra -I../erpc_c/port -I../erpc_c/transports -I../erpc_c/setup  -I../erpc_c/config -c server_main.c  -o server_main.o -ggdb

4. Link both of these into a server executable which actually runs the server
# Cmd to compile server all 
g++ ../Debug/Darwin/erpc/obj/erpc_c/infra/*.o  ../Debug/Darwin/erpc/obj/erpc_c/setup/*.o ../Debug/Darwin/erpc/obj/erpc_c/port/*.o ../Debug/Darwin/erpc/obj/erpc_c/transports/*.o server.o server_main.o -o server -ggdb

Client-Side Commands

1. Compile client.o made from erpc_matrix_multiply_client.cpp

g++ -I../erpc_c/infra -I../erpc_c/port -I../erpc_c/transports -I../erpc_c/setup  -I../erpc_c/config -c service/erpc_matrix_multiply/erpc_matrix_multiply_client.cpp -o client.o -ggdb

2. Compile client_main.c

# Cmd to compile client main 
gcc -I../erpc_c/infra -I../erpc_c/port -I../erpc_c/transports -I../erpc_c/setup  -I../erpc_c/config -c client_main.c  -o client_main.o -ggdb

# Cmd to compile client all 
g++ ../Debug/Darwin/erpc/obj/erpc_c/infra/*.o  ../Debug/Darwin/erpc/obj/erpc_c/setup/*.o ../Debug/Darwin/erpc/obj/erpc_c/port/*.o ../Debug/Darwin/erpc/obj/erpc_c/transports/*.o client.o client_main.o -o client -ggdb`

Am I doing something

MichalPrincNXP, 2020-03-13:
Hi @hale2bopp 
the only thing I have noticed when looking into your code is the server_main.cpp: the erpc_server_run() should not be called in the while loop as described in https://github.com/EmbeddedRPC/erpc/wiki/eRPC-infrastructure 
It should be like this:
```
erpc_server_run();
while (1)
    {
    }
```
I recommend to have a look at https://github.com/EmbeddedRPC/erpc/blob/master/test/common/unit_test_tcp_client.cpp and https://github.com/EmbeddedRPC/erpc/blob/master/test/common/unit_test_tcp_server.cpp sources: these are similar to your case but without using the setup functions.

hale2bopp, 2020-03-25:
Issue was - client runs as a server by default, since the boolean which sets it as a client or as a server is set to 1 by default.

closed 2020-03-25 by hale2bopp