EmbeddedRPC/erpc #93: Arbitrated client stucks -> dead lock

willmann9527, 2020-04-24:
Condition: 
This bug occurres if a message will be lost or the server is not yet available.

Issue: 
If the client invokes a procedure, erpc sends the invoke out no matter if there is someone, Right depends on the driver. After that, the arbitrated transport waits forever for a reply but this reply will never received and the process will stuck -> dead lock. As well when the server is coming up.  

Solution:
Replace the following timeout kWaitForever by a member variable mu32_ReceiveTimeout, which is going to be initialized in the constructor and can be set by a setter method.
info->m_sem.get(Semaphore::kWaitForever);

BTW: When are you going to release a new version?

MichalPrincNXP, 2020-04-27:
Hello @willmann9527 , thank you for reporting this issue. Agree, this needs to be solved. BTW, what transport layer are you using, is it MU_transport? As for the next planned release it should be available in June.

willmann9527, 2020-04-27:
Hello @MichalPrincNXP, 
thank you for the rapid reply. 
I use my own transport layer due to the fact that a similar to IP layer is between eRPC and the physical layer.
However, how do you handle a communication from one central point to two communication partners? The thing is, p_client is generated by the erpc generator and must be reassigned before we call an erpc procedure, which results in we have to write wrappers for the generated procedures to set this pointer p_client. I guess you have also a use case where you have two clients in one application and hopefully a smarter solution.

MichalPrincNXP, 2020-04-27:
Hello @willmann9527 , if I understand your case correctly, you would like to have one eRPC client to communicate with two independent eRPC servers, correct? Then you have two transports, each for one client-server communication, and there is not necessary to have the transport arbitrator. In fact there are two independent eRPC client instances running on the client side then.

willmann9527, 2020-04-27:
Hello @MichalPrincNXP, 
Yes, right. How do you implement these two independent eRPC client instances running on the client side as the generated code uses just one client pointer p_client?

MichalPrincNXP, 2020-04-27:
Yes, you are right, this is the corner case that is not handled by the generator. Unfortunately, it is up to the user to solve now (by a script or a wrapper like you did it).

RamseyTI, 2020-04-27:
Our main use-case requires multiple client-server instances.  To achieve this, I added instance versions of all setup functions; they don't use the static global objects. I changed the generator to create a client specific name for p_client. In the setup code, I bind the client to the client manager instance. I'm curious, if you share the same API with two instances, how do you know which instance to bind p_client with? Is there some "higher entity" which knows this? Or do you have two different components, which each happen to use the same API but on different remote connections?

willmann9527, 2020-04-28:
Hi @RamseyTI, 
we are planning to call a procedure in front of each erpc call, which sets the global client p_client to the proper remote connection.
We also considered your solution but we wanted to stay as keep as possible at the erpc repository to facilitate updates from it. Frankly speaking, if the generator is capable of taking a client parameter would be the best solution for us. Then, the setup function uses a list of clients to initialize these clients.

RamseyTI, 2020-04-28:
@willmann9527, understood. Thanks for the reply.

willmann9527, 2020-05-07:
Hi @RamseyTI,
may I have your tailored generator (solution and/or exe) and the setup files to check if it would be work for me?
That would be awesome. Thank you in advance.

RamseyTI, 2020-05-08:
Hello @willmann9527 
There are three elements to our object binding solution:
- Modify client source template to replace global symbol with module specific name- Add setup functions which take object references
- Setup code
-

anirudhgargi, 2020-07-02:
@willmann9527 
For the issue of client getting stuck, in a way a eRPC client timeout for each eRPC call. I am also concerned about this same issue in my implementation. Need a mechanism in client to unblock the execution flow. 

Can you share how you attempted this - maybe that sample setter function (method) code and Sem variable handling ? 

Talking about this idea - 
"Solution:
Replace the following timeout kWaitForever by a member variable mu32_ReceiveTimeout, which is going to be initialized in the constructor and can be set by a setter method.
info->m_sem.get(Semaphore::kWaitForever);"

Thanks.

willmann9527, 2020-07-02:
@anirudhgargi 
Frankly speaking, in the end I decided against the arbitrater. But it should be not that hard. Just use a member, e.g. mu32_Timeout, and set it in the constructor to Semaphore::kWaitForever. Then set up a setter procedure, for instance SetTimeout(uint32 u32_InTimeout) { mu32_Timeout = u32_InTimeout; }. Last but not least, use the member mu32_Timeout instead of the Semaphore::kWaitForever: info->m_sem.get(mu32_Timeout);.
Was that helpful?

anirudhgargi, 2020-07-06:
@willmann9527  Thanks for the prompt reply. Let me implement and validate. Will share feedback soon.

pgu-swir, 2020-08-26:
I think it would be better to release pending clients in erpc_transport_arbitrator::receive with something like
```
        erpc_status_t err = m_sharedTransport->receive(message);
        if (err)
        {
		// if we timeout, we must unblock all pending client(s)	
		if (err == kErpcStatus_Timeout)
		{	
			PendingClientInfo *client = m_clientList;
			for (; client; client = client->m_next)
			{
				if (client->m_isValid)
				{
					client->m_sem.put();
				}	
			}
		}
               return err;
        }
```

Hadatko, 2022-04-25:
Hi @pgu-swir @MichalPrincNXP 
do you think this jira can be closed based on merged PR?

MichalPrincNXP, 2022-04-26:
I would close it if others involved agree too. @willmann9527 @RamseyTI @anirudhgargi  any comments?

Hadatko, 2023-09-27:
Closing for inactivity.

closed 2023-09-27 by Hadatko