EmbeddedRPC/erpc #94: Server Client implementation on Linux Over UART

anirudhgargi, 2020-04-26:
Hello guys, 

Need some help in erpc Client and Server application development. 

I am trying to evaluate erpc librart and for that i am building a erpc client and server on Linux (C/C++ code base). Which shall talk over a UART interface given the circumstances, yeah i know, in next release i have IP support i can then use TCP as the transport type. 
I was able to compile a basic client and server application integrating the erpc_c files. I want to use the serial transport type already there in the transports folder. I believe it works ok ? 

The issue i am facing is that from the demo client app, when i send a actual function request (erpc call) , it seems nothing is sent on the UART interface. Is something missing in the transport adaption ? 

For development time i made 2 virtual tty interfaces - say _/dev/pts/1_ and _dev/pts/2_ which talk to each other (i used _socat_ utility to achieve this). So anything i write to _pts/1_ goes to other side and vice versa. This is working ok, i checked over a serial terminal app (_minicom_). In a way stimulating a UART interface on same machine. 

But, when i send the my 1st erpc client request over to server, i do not see anything on UART interface, i am sniffing that interface constantly. 

My **server** app code looks like this: 

  1 #include <cstdio>
  2 #include "erpc_server_setup.h"
  3 #include "erpc_transport_setup.h"
  4 #include "erpc_transport.h"
  5 #include "erpc_mbf_setup.h"
  6 
  7 #include "initializelib.h"
  8 #include "initializelib_server.h"
  9 
 10 /* implementation of function call */
 11 void C_Initialize(void)
 12 {   
 13     printf("C_Initialize ++ Server side \n");
 14     printf("C_Initialize -- Server side \n");
 15 }
 16 
 17 int main()
 18 {
 19     /* Init eRPC server environment */
 20     /* UART transport layer initialization */
 21     erpc_transport_t transport = erpc_transport_serial_init("/dev/pts/1", 115200);
 22     
 23     printf("Server Uart Init ok...\n");
 24     /* MessageBufferFactory initialization */
 25     erpc_mbf_t message_buffer_factory = erpc_mbf_dynamic_init();
 26     
 27     /* eRPC server side initialization */
 28     erpc_server_init(transport, message_buffer_factory);
 29     
 30     /* connect generated service into server, look into erpc_matrix_multiply_server.h */
 31     erpc_service_t service = create_initializelibService_service();
 32     erpc_add_service_to_server(service);
 33     
 34     printf("Server Init done...\n");
 35     
 36     /* run server */
 37     printf("Waiting for client calls...\n");
 38     //    erpc_server_run(); /* or erpc_server_poll(); */
 39     
 40     /* run server on poll */
 41     while (1)
 42     {
 43         /* Process eRPC requests */ 45 
 46         // To Do
 47         /* handle error status */
 48         //if (status != kErpcStatus_Success)
 49         //{
 50         //  /* print error description */
 51         //  erpc_error_handler(status, 0);
 52         }
 53 
 54     printf("Server Running done...\n");
 55     return 0;
 56 
 57     //server deinit here next
 58 
 59 }

My **client** code looks like this:

 1 /* Client code */"
  2 #include <cstdio>
  3 #include "initializelib.h"
  4 #include "erpc_client_setup.h"
  5 #include "erpc_transport_setup.h"
  6 #include "erpc_transport.h"
  7 #include "erpc_mbf_setup.h"
  8 
  9 int main()
 10 {
 11     /* Init eRPC client environment */
 12     /* UART transport layer initialization */
 13     erpc_transport_t transport = erpc_transport_serial_init("/dev/pts/2", 115200);
 14     
 15     /* MessageBufferFactory initialization */
 16     erpc_mbf_t message_buffer_factory = erpc_mbf_dynamic_init();
 17     
 18     /* eRPC client side initialization */
 19     erpc_client_init(transport, message_buffer_factory);
 20     
 21     /* other code like init matrix1 and matrix2 values */
 22     printf("Client init ok... \n");
 23     
 24     /* call eRPC functions */
 25     printf("Calling C_initalize function ... \n");
 26     void C_Initialize(void);
 27 
 28     /* other code like print result matrix */
 29     printf("Control back in client side ... \n");
 30     
 31     return 0;
 32 } 

Can you share some insights, why possibly the client is not able to talk to the server over UART in this setup ?

anirudhgargi, 2020-04-26:
Got it working. Please ignore. 

Found couple of issues related to UART initialization.

closed 2020-04-26 by anirudhgargi

kumary-vvdn, 2022-01-17:
Hi,
I am able to call services in server from client. I have a need to send data from server to client without a request from client. Is there a way to implement it?. Please provide your suggestion.

Hadatko, 2022-01-17:
Hi @kumary-vvdn , you need use arbitrator layer. Something is written in documentation, something in issues here on github, you may also look into test for example test_arbitrator.

Hadatko, 2022-01-17:
@anirudhgargi Based on your response i am closing this issue. We are happy that you made it working ;)