EmbeddedRPC/erpc #95: run erpc server task in multi threaded environment

tmordeko, 2020-05-04:
Hi,

Running with FreeRTOS OS, erpc_server_run() is called from from separate task.
The way previous function is implemented is in endless loop, never giving up on cpu waiting for ever for received data on the transport layer.

Is there a way to optimize that code such that it will be waiting on events from the transport layer ?

Thanks,
Tzvika

Hadatko, 2020-05-11:
Hi @tmordeko , di you consider poll function instead of run?

tmordeko, 2020-05-12:
Hi @Hadatko,

I am aware of the poll mode, but it basically brings me back to the same problem as I am waiting on the uart channel.

Tzvika

Hadatko, 2020-05-12:
I see. Looks like you need another implementation of uart receive function. For example erpc could have possibility to have ring buffers and save messages on background (something like rpmsg).

tmordeko, 2020-05-12:
is it something supported today or I should implement it on my own ?

MichalPrincNXP, 2020-07-14:
Hi @tmordeko , have you coped with this problem? Any update?

Hadatko, 2020-09-29:
@tmordeko today i was talking with @MichalPrincNXP about this issue. It looks like solution is simple to add erpc_thread.h and implement take/put sempahores based on that header file. Take semaphore instead of while loop and put semaphore from interrupt once data copied. Do you think you can implement this feature these days? If we will do it quickly it will show up in release next release (1.7.5?).

tmordeko, 2020-09-30:
Hi @Hadatko ,

Thanks for the update, 

When is next release1.7.5 expected to be published ?

Can you do the change and send me the CR so we will integrate it ?

/Tzvika

Hadatko, 2020-09-30:
If you are not interested in doing change i will let it on @MichalPrincNXP.

MichalPrincNXP, 2020-09-30:
Hello, this seems to me like important and easy to fix issue. I would try to do my best to update the uart transport layer by the end of next week when the internal code freeze date is planned. Will provide you the patch later, ok?

Hadatko, 2020-09-30:
@MichalPrincNXP just thinking what should be best way of developing this framework from NXP site. I think the new features could be synced back to develop on github once they are pushed into internal repository. So community can see progress and what will be in next release and they will have also access to the latest code.

If we would have bigger active users, then NXP could also do PR to github to get review from community. Just thinking....

tmordeko, 2020-09-30:
Hi @MichalPrincNXP 

Thanks a bunch !

MichalPrincNXP, 2020-09-30:
> @MichalPrincNXP just thinking what should be best way of developing this framework from NXP site. I think the new features could be synced back to develop on github once they are pushed into [...]

yes, agree, we should get back to the monthly update concept, but ... it is not easy to manage all that ...

MichalPrincNXP, 2020-10-02:
Hello @tmordeko ,
attached is the updated CMSIS UART transport layer ... could you please try on your side and let me know if it helps to solve your problem? I have made quick testing on an erpc example that runs FreeRTOS-based client on the embedded side and the erpc server on the PC side, using this updated CMSIS UART transport, and it seems to be working fine. Thanks.

Note1: erpc_threading.h and erpc_threading_freertos.cpp needs to be added into your project if not already done.
Note2: UART interrupt service priority needs to be set correctly in FreeRTOS env. otherwise assert could be reached (if enabled), see portASSERT_IF_INTERRUPT_PRIORITY_INVALID. One could use the NVIC_SetPriority() function for this purpose, for instance: NVIC_SetPriority(FLEXCOMM0_IRQn, 5);
[transports.zip](https://github.com/EmbeddedRPC/erpc/files/5316576/transports.zip)

Hadatko, 2020-10-04:
@MichalPrincNXP it would be fine in future to have PR instead of zip file. It is better way to comment changes from various od peaople and easy merge process + history...

Hadatko, 2021-07-22:
I think this is already merged, am i correct? Can be this done also for other transports like e.g.: MU???

MichalPrincNXP, 2021-07-26:
Yes, it has been integrated already, thanks for the notice, I am closing now ...

closed 2021-07-26 by MichalPrincNXP