EmbeddedRPC/erpc #96: Serial eRPC on macOS takes a long time (10secs) with `ret = read(fd, buf + len, size - len);`

andrew-mclachlan, 2020-05-08:
More a question than a bug.

I'm setting up an eRPC serial connection on macOS with `/dev/cu.usbmodem0001A00000001` and baud rate `921600`.
I am finding that the server is receiving and sending bytes find but the read is taking a small amount of time compared with Linux and Windows.
In other words, `ret = read(fd, buf + len, size - len);` takes about 10 seconds to complete in some cases. When compared to Linux where the read is instantaneous.

Any thoughts?

andrew-mclachlan, 2020-06-20:
This could be a mismatch between eRPC structures on device Vs. host. Need to debug to check.

MichalPrincNXP, 2020-07-14:
Hi @andrew-mclachlan , is this issue still valid or have you already solved that? I am sorry not being to help you with macOS.

andrew-mclachlan, 2020-07-14:
Unfortunately I have not been able to spend time on debugging more. Hoping to do that shortly.
Wondered if anyone else had seen similar.

MichalPrincNXP, 2020-07-15:
Hi @andrew-mclachlan , from previous posts I know @hale2bopp is working with erpc on MacOS (see #88 , #90 ). Maybe she could help you ...

Hadatko, 2023-06-06:
Maybe can be fixed by #183