EmbeddedRPC/erpc #97: Weak error description when handling error on server side

Hadatko, 2020-05-17:
```
def handle_invocation(self, methodId, sequence, codec):
        try:
            self._methods[methodId](sequence, codec)
        except KeyError:
            raise RequestError("invalid method ID (%d)" % (methodId))
```

Easy fix is change error message to ("invalid method ID  (%d) or implementation failed"% (methodId))

But this can be confusing for developers (catch theirs errors). I think could option to have could be:
on same place as self._methods is implemented we can have (generated) attribute self._ids = [.....]
This will generate array of ids and we can do this:

```
def handle_invocation(self, methodId, sequence, codec):
     if methodId not in  self._ids:
            raise RequestError("invalid method ID (%d)" % (methodId))        
     self._methods[methodId](sequence, codec)
```

MichalPrincNXP, 2020-07-14:
@Hadatko , do you plan to create PR?

Hadatko, 2020-09-25:
I will ;)

Hadatko, 2020-09-29:
Merged

closed 2020-09-29 by Hadatko