EmbeddedRPC/erpc #101: Callbacks not doing what they should

jpdoyon, 2020-06-11:
My understanding of what a callback should be doing in the context of RPC is the following:

[image]

But what I get from eRPC callback test:

[image]

Is there a way to get an asynchronous callback from the server to the client?  I can't figure out how to do that...

Hadatko, 2020-06-12:
Hi @jpdoyon Both described case should work well. It depends on which side you will include erpc myCallback1 client implementation and where you will implement server implementation. In first case you need client and server on both sides and use transportarbitrator.

In first case you need mycallback1 client erpc implementation on server side of your picture and call it. Server implementation of this call you need on Client side of you image. 

This way both picture are valid at the same time only change would be that myCallback1 call an implementation on same side would be on client side of your image.

Hadatko, 2020-06-12:
[image]

jpdoyon, 2020-06-15:
I must admit I'm kinda new to eRPC, so if I understand correctly, for option A, I need to have two .erpc files?  In that case, I can set the pointer to myCallback1 from the client (it can be any function that the server doesn't necessarily know about upfront)?

Do you have examples of using the transport arbitrator?

Hadatko, 2020-06-16:
One erpc file is enough. You need two groups of declarations.
Example is here: https://github.com/EmbeddedRPC/erpc/tree/develop/test/test_arbitrator. 

Wel for safety reason, you can't assign any function. The function need be declared in erpc file as callback function. So the definition exists on both sides (user and erpc implementation).

Hadatko, 2020-06-16:
Exemple with callbacks here: https://github.com/EmbeddedRPC/erpc/tree/develop/test/test_callbacks

jpdoyon, 2020-06-16:
Great!  I actually had found the test_arbitrator yesterday and started playing with it.  I actually got it to do what I want this morning!  So thank you so much for your invaluable help!

closed 2020-06-16 by jpdoyon